Duet3D/RepRapFirmware #95: Tabbing Between Fields in DWC Causes Java Error

bmalcheski, 2017-05-05:
I am currently running DWC 1.15b but I believe this started on 1.15a. I get the below error when I tab between fields (ie temperature inputs) in the DWC. Currently running FW 1.18.1 and server 1.03(ch fork)

>  JavaScript Error

A JavaScript error has occurred so the web interface has closed the connection to your board. It is recommended to reload the web interface now. If this happens again, please contact the author and share this error message:

Message: Uncaught ReferenceError: windowXsSm is not defined
URL: http://10.0.0.57/js/dwc.js
Line: 74187:38
Error object: {}

dc42, 2017-05-05:
This is not an issue with RRF, it is an issue with either your browser or with Duet Web Control. Please report it at https://github.com/chrishamm/DuetWebControl/issues stating which browser and browser version you are using.

closed 2017-05-05 by dc42

bmalcheski, 2017-05-05:
Sorry. Wasn't sure where to put it. Thanks for the link.