Duet3D/RepRapFirmware #96: RepRapFirmware for Alligator

MarcoAntonini, 2017-05-12:
Hi David,

I'm playing with this amazing firmware on an Alligator board, I would like to take advantage of a network connection :)
Alligator has different ethernet PHY chip but it is compatible with Duet, I'm testing it, it's working properly!

The only problem encountered is that the Alligator v2 pins Mapping  does not allow a hardware PWM on heaters.  I need to implement PWM with a timer counter, do you have some advice to not to disturb more sensitive timing operations  like stepper?

Thank you
Marco

dc42, 2017-05-12:
Do you mean that the heater mosfets are controlled by pins attached to timer/counter outputs, or that you need to use a timer/counter to do software PWM? Even though the older Duets have all the heaters driven from hardware PWM pins, the CoreNG hardware abstraction layer supports using timer/counter output pins for PWM as well.

jamesarm97, 2017-05-12:
I have an Alligator board sitting around for over a year or two it seems. Watching this one ;) I would love to get it into service but not having ethernet was the show stopper for me back in the day. @MarcoAntonini  anything special you had to do to compile for Alligator?

MarcoAntonini, 2017-05-12:
@dc42 Unfortunately I need timer/counter for software PWM, my pins not connected to a TC channel, it cannot received PWM .

@jamesarm97 , I'm sorry, an old dc42 fork exists for alligator but the sources are lost, I would like to start with this 1.19 version

dc42, 2017-05-12:
RRF for the Duet 0.6/0.8.5 (which is closer to the Alligator than the later Duets) uses 2 timer channels, one for step pulse timing and one for the network tick. A block at the end of the Pins_Duet.h file defines which timers are used for this function. Atmel's nomenclature for timers is a little strange, so although the timers are channels 0 and 1 of the same timer (TC1), the interrupts IDs are TC3 and TC4. You can pick any of the remaining 7 timer channels to do your software PWM i the timers are not being used to drive heater or fan outputs using barware PWM. Set the interrupt priority in the NVIC for your chosen timer to be worse (higher number) than the TC3 interrupt priority.

However, I've just looked at the schematic for the Alligator at https://github.com/3Dartists/AlligatorBoard/blob/master/SCHEMATICS/SCH_ALLIGATOR-BOARD-REV02.pdf and I see that the bed heater driver and all the extruder heater drivers except HE_HEAT_EN1 are driven from PWM-capable pins. So it is only HE_HEAT_EN1 that needs software PWM. You might consider modding the board so that the on-board extruder heater driver is driven from HE_HEAT_EN2 instead, then you wouldn't need to use software PWM at all. The FAN_EN1 signal is driven from a pin attached to a timer/counter, so that one is OK too.

HTH David

MarcoAntonini, 2017-05-12:
Effectively swapping hot-end1 with bed can solve the problem.
Need only one pwm software for the second fan, I will use one of the available free timers.
Thanks for the information
Marco

closed 2017-05-12 by MarcoAntonini

MarcoAntonini, 2017-05-13:
I am editing Alligator variants but according to atmel datasheet it seems that FAN_EN1 will not receive PWM hardware,  FAN_EN1 connected to PA5  which is a PWMFIx , defined as Pwm Fault Input signal.
Ok, I will implement fan pwm software in one of the free timers.

dc42, 2017-05-13:
FAN_EN1 is connected to PA5 which has alternate function TIOA2. So you could use hardware PWM generated by that TC, just as the Duet WiFi does.

MarcoAntonini, 2017-05-13:
Ok, switched PA5 on A peripheral , it work. Thank yuo

MarcoAntonini, 2017-05-23:
Finally I physically tested Alligator r2 on a delta printer with dc42 RepRapFirmware, It seems to work very well!

No pwm software is used:
 - Mosfets heater 1,2,3,4  are in hardware PWM
 - Fan1 is hardware PWM
 - Fan2 is mapped as always on or alway off
 - Mosfets Bed is in bang-bang

Sources available in [my dc42 RepRapFirmware fork](https://github.com/MarcoAntonini/RepRapFirmware)
The [dc42 RepRapFirmware 1.19beta1 release fo Alligator](https://github.com/MarcoAntonini/RepRapFirmware/releases) is available! 

Special thanks to David for help and support.

dc42, 2017-05-23:
Greqt! You might like to announce it at http://forums.reprap.org/list.php?13.

jamesarm97, 2017-05-23:
Great. Can't wait to finally put my Alligator to use. Almost like having a new Duet with Ethernet.

dc42, 2017-05-24:
I'll merge your change into my fork, probably in time for the next release. Assuming that you haven't made substantial changes to the networking code apart from for the interface to your PHY chip, then I think you have a lot of files in your Alligator folder that are identical (or nearly so) to ones in the Duet folder. In which case, we ought to split these out into a separate folder.

MarcoAntonini, 2017-05-24:
David, this is a great news :)
Yes, networking library files are duplicated, they can be shared with Duet , split it into a separate folder is 
much more orderly.