Duet3D/RepRapFirmware #97: SD CARD Hot-Plug

Laiwxn, 2017-05-13:
Hi David,I found that the On-Board SD Card didn't support Hot Plug(Duet 0.8.5).If I Pulg the SD Card,I must Reset the board.But I have checked the schematic and found that  the Card Detect Pin has been tied to PB27,it means this board can support the Hot Pulg.Is this a bug or anything else?

dc42, 2017-05-13:
1. After re-inserting the card, you can send M21 to get the board to re-initialise the card instead of resetting the board. There is a firmware bug that means this doesn't always recognise some changes to the files on the card, but this is fixed in firmware 1.19alpha2 and later.

2. There is a bug in the ATSAM3X8E chip that means that once we enable the MCU temperature sensor, the CD pin input no longer works. So we wouldn't be able to support ho-plugging on the Duet 0.6/0.8.5 without using M21 unless we disable MCU temperature monitoring. This doesn't apply to the Duet WiFi and Duet Ethernet, so we may support hot-plugging on those boards in the future.

Laiwxn, 2017-05-16:
Got it,thank you！

closed 2017-05-16 by Laiwxn