Duet3D/RepRapFirmware #99: FTP connexion

Lepterois, 2017-05-14:
Hi!
I've done the firmware update to the 1.18.1 on my Ormerod2. Since this update I cannot connect anymore to my Ormerod in FTP using Filezilla.
What can I do for this? before I used Version 1.09 and it was working fine. What is changing between the 2 versions?

Thanks in advance

dc42, 2017-05-14:
Read the upgrade notes in the WHATS_NEW file in github.

closed 2017-05-14 by dc42

Lepterois, 2017-05-14:
thanks for reply.
the history I found stop at 1.16 version and I was at version 1.09 ;-)
and that's like chinese for me lol
but I get the answer somewhere else
regards

dc42, 2017-05-14:
https://github.com/dc42/RepRapFirmware/blob/dev/WHATS_NEW.md

Lepterois, 2017-05-14:
Thanks very much.
this link is very interesting!