Duet3D/RepRapFirmware #100: Firmware name renaming error in web upload

MarcoAntonini, 2017-05-17:
Hi David,

I have noticed a small error in firmware upload via web interface:

I set the firmware name of the Alligator variant as **RepRapFirmware-Alligator.bin** in iap.h and in Pins_Alligator.h

`const char * const defaultFwFile = "0:/sys/RepRapFirmware-Alligator.bin";
`

`#define IAP_FIRMWARE_FILE "RepRapFirmware-Alligator.bin"
`

 When the new firmware is loaded via web interface, automatically javascript code renames it in RepRapFirmware.bin .
This causes the M997 command to fail because "RepRapFirmware-Alligator.bin" files will not be found

```
...
if (uploadFirmwareFile.toUpperCase() != "REPRAPFIRMWARE.BIN")
	{
		// The firmware filename is hardcoded in the IAP binary, so try to rename the uploaded file first
		$.ajax("rr_move?old=" + encodeURIComponent("/sys/" + uploadFirmwareFile) + "&new=" + encodeURIComponent("/sys/RepRapFirmware.bin"), {
			dataType: "json",
...
```

dc42, 2017-05-17:
Hi, you will have to ask chrishamm if he is prepared to change DuetWebControl to accommodate the Alligator board. It already. The github repo for that project is https://github.com/chrishamm/DuetWebControl.

The comment in the JavaScript about the firmware filename being hard coded in IAP is no longer entirely correct, because the filename is passed to IAP in recent versions, although there is a hard-coded default in case the firmware doesn't pass a filename,

MarcoAntonini, 2017-05-18:
Ok, for now leave the firmware name as the original RepRapFirmware.bin. 
Very useful to upload firmware in web interface!

thank you
Marco

closed 2017-05-18 by MarcoAntonini