Duet3D/RepRapFirmware #101: Feature Request: M204 S Parameter

notnyt, 2017-05-23:
Marlin and other firmware take the S parameter to set normal move speeds.  Many slicers (cura) use this to control acceleration for different portions of the print.  Is it possible to support this?  If you need I'll fork, patch, and submit a pull request.

dc42, 2017-05-23:
RRF implements M204 as described on the reprap,org wiki as follows:

"Marlin notes: After Mar11-2015, the M204 options have changed in Marlin:
P = Printing moves
R = Retract only (no X, Y, Z) moves
T = Travel (non printing) moves
The command M204 P800 T3000 R9000 sets the acceleration for printing movements to 800mm/s^2, for travels to 3000mm/s^2 and for retracts to 9000mm/s^2."

I left out the R parameter because it seems to be pointless, as extruder acceleration is set using M201.

notnyt, 2017-05-23:
Ah, I see in Marlin:

  if (code_seen('S')) {  // Kept for legacy compatibility. Should NOT BE USED for new developments.
    planner.travel_acceleration = planner.acceleration = code_value_linear_units();           SERIAL_ECHOLNPAIR("Setting Print and Travel Acceleration: ", planner.acceleration);
  }

dc42, 2017-05-28:
Implemented in version 1.19 beta 2.

closed 2017-05-28 by notnyt