Duet3D/RepRapFirmware #105: U axis does not move (1.19beta1 DuetWifi)

Alex9779, 2017-05-28:
Hi,
as my rebuild of the box is nearly finished I tested the drives today with the newest firmwares.
X, Y and Z work but U does not.
I checked the cables and all is fine.
If I connect the U motor to the X port then the motor moves when I move X.
If I remap the motors different then the U motor moves (mapping motor 0 to U and 3 to X) and X does not, so the drivers are also ok.
Here is my config:
<details>
<summary>Config</summary>

```
; General preferences
M111 S0 ; Debugging off
G21 ; Work in millimetres
G90 ; Send absolute coordinates...
M83 ; ...but relative extruder moves
M555 P2 ; Set firmware compatibility to look like Marlin

; Network
M550 PBigBox ; Set machine name
M540 P0x60:0x01:0x94:0x0C:0x54:0x28 ; set MAC address
M552 P0.0.0.0 S1 ; Enable network and acquire dynamic address via DHCP
M554 P172.16.0.1 ; Set gateway

; Drives
M569 P0 S1 ; Drive 0 goes backwards (X)
M569 P1 S1 ; Drive 1 goes forwards (Y)
M569 P2 S1 ; Drive 2 goes forwards (Z)
M569 P3 S1 ; Drive 3 goes forwards (U)
M569 P5 S0 ; Drive 5 goes backwards (E0)
M569 P6 S0 ; Drive 6 goes backwards (E1)

; Configure U axes
M584 X0 Y1 Z2 U3 E5:6

; Movement
M906 X560 Y1300 Z1200 U560 E600 I60 ; Set motor currents (mA) and motor idle factor in per cent (tweaked for motor temps)
M201 X400 Y400 Z100 U400 E10000:10000 ; Set accelerations (mm/s^2)
M203 X9000 Y9000 Z360 U9000 E3000:3000 ; Set maximum speeds (mm/min)
M566 X480 Y480 Z24 U480 E300:300 ; Set maximum instantaneous speed changes (mm/min)
M84 S30 ; Set idle timeout

; Stepping
M350 X32 Y16 Z16 U32 E16:16 I1 ; Configure microstepping with interpolation
M92 X160 Y360 Z1600 U160 E418.5:418.5 ; Set steps per mm

; Set limits
M208 X-16.5 Y-42.3 Z0 U62.5 S1 ; Set axis minima
M208 X273 Y196 Z300 U349 S0 ; Set axis maxima

; Endstops
M574 X1 Y1 Z0 U2 S1 ; Define active high microswitches

; Z probe
;M558 P1 X0 Y0 Z1 U0 H3 F120 T3000 ; Set Z probe type to unmodulated, the axes for which it is used and the probe + travel speeds (IR Probe)
;G31 P500 X-8.8 Y60 Z0 ; Set Z probe trigger value, offset and trigger height (IR Probe)
M558 P5 X0 Y0 Z1 U0 H2 F10 T1500 ; Set Z probe type to unmodulated, the axes for which it is used and the probe + travel speeds (BLTouch)
G31 P1 X-20.6 Y69.88 Z1.45 ; Set Z probe trigger value, offset and trigger height (BLTouch)

; Heaters
M143 H0 S120 ; Set maximum heater temperature to 120C
M143 H1 S320 ; Set maximum heater temperature to 320C
M143 H2 S320 ; Set maximum heater temperature to 320C
M307 H0 A164.0 C545.5 D1.0 B0
M307 H1 A474.8 C236.9 D7.0 B0
M307 H2 A474.8 C236.9 D7.0 B0
M305 P0 T100000 B4388 C0.0000000000000000 R4700 L0 H30 ; Set thermistor + ADC parameters for heater 0
M305 P1 X200 ; Set thermistor + ADC parameters for heater 1 and remap it to channel 200
M305 P2 X201 ; Set thermistor + ADC parameters for heater 2 and remap it to channel 201

M307 H3 A-1 C-1 D-1 ;
M307 H4 A-1 C-1 D-1 ;
M307 H5 A-1 C-1 D-1 ;
M307 H6 A-1 C-1 D-1 ;
M307 H7 A-1 C-1 D-1 ;

M143 S320 ; Set maximum heater temperature to 320C

; Fans
M106 P0 S0 I0 F10 H-1 B0.25 L30 ; Set fan 0 value, PWM signal inversion and frequency. Thermostatic control is turned off (Print fan left)
M106 P1 S1 I0 F10 H1 T40 ; Set fan 0 value, PWM signal inversion and frequency. Thermostatic control is turned on (Hotend fan left)
M106 P2 S1 I0 F10 H2 T40 ; Set fan 1 value, PWM signal inversion and frequency. Thermostatic control is turned on (Hotend fan right)
M106 P3 S0 I0 F10 H-1 B0.25 L30 ; Set fan 0 value, PWM signal inversion and frequency. Thermostatic control is turned off (Print fan right)
;M106 P6 S0.5 I0 F5 ; Set fan 6 value, PWM signal inversion and frequency. (Cover fan)
M106 P7 S1 I0 ; LEDs

; Grid
;M557 X5:270 Y35:195 S10
M557 X5:270 Y35:195 S15

; Tools

; Printing tools
; Define tool 0
; Default left hotend
M563 P0 D0 H1 X0 F0:3
G10 P0 X0 Y0 Z0 ; Set tool 0 axis offsets
G10 P0 R0 S0 ; Set initial tool 0 active and standby temperatures to 0C

 ; Define tool 1
; Default right hotend
M563 P1 D1 H2 X3 F0:3
G10 P1 U0.1 Y-0.1 Z0.1 ; Set tool 1 axis offsets
G10 P1 R0 S0 ; Set initial tool 1 active and standby temperatures to 0C

; Define tool 5
; Create a tool that prints 2 copies of the object using both carriages
M563 P5 D0:1 H1:2 X0:3 F0:3 ; tool 5 uses both extruders and hot end heaters and maps X to both X and U
G10 P5 X50 Y0 U-50 S0 R0 ; set tool offsets and temperatures
M567 P5 E1:1 ; set mix ratio 100% on both extruders
M568 P5 S1 ; turn on mixing for tool 10

; Supporting tools for script usage
; Define tool 10
; Left hotend unscripted
M563 P10 D0 H1 X0 F0:3
G10 P10 X0 Y0 Z0 ; Set tool 10 axis offsets
G10 P10 R0 S0 ; Set initial tool 10 active and standby temperatures to 0C

; Define tool 11
; Right hotend unscripted
M563 P11 D1 H2 X3 F0:3
G10 P11 U-0.6 Y-0.2 Z0.2 ; Set tool 11 axis offsets
G10 P11 R0 S0 ; Set initial tool 11 active and standby temperatures to 0C

; Define tool 15
; Create a tool that prints 2 copies of the object using both carriages (unscripted)
M563 P15 D0:1 H1:2 X0:3 F0:3 ; tool 15 uses both extruders and hot end heaters and maps X to both X and U
G10 P15 X50 Y0 U-50 S0 R0 ; set tool offsets and temperatures
M567 P15 E1:1 ; set mix ratio 100% on both extruders
M568 P15 S1 ; turn on mixing for tool 10

; Define tool 99
;Probe tool IR Probe
;M563 P99 X0
;G10 P99 X-8.8 Y60 Z0; Set tool 99 axis offsets

; Define tool 99
;Probe tool BLTouch
M563 P99 X0
G10 P99 X-20.6 Y69.88 Z0; Set tool 99 axis offsets

M280 P4 S35
M280 P5 S125
```

</details>

dc42, 2017-05-28:
I've just tested U axis movement on 1.19beta2 which should be released later today, and it appears to be working. So please test again with 1.19beta2 when it is available.

Alex9779, 2017-05-28:
Ok with 1.19beta2 it works... thanks...

closed 2017-05-28 by Alex9779

dc42, 2017-05-28:
Thanks for confirmation!