Duet3D/RepRapFirmware #107: Thermostatic fan control not working for H101 TMC2660

notnyt, 2017-05-29:
M106 P2 T60 H101  ends up always leaving the fans on, even when the stepper drivers are cold.  Set to T600 just to be sure, same thing.  Works fine with CPU (H100).

M106 P2
Fan2 frequency: 250Hz, speed: 100%, min: 10%, blip: 0.10, inverted: no, trigger: 60C, heaters: 101

notnyt, 2017-05-29:
Is the TMC2660 temperature only reporting as 0, 100, or 150?

notnyt, 2017-05-29:
11.3 Overtemperature Detection
The TMC2660 integrates a two-level temperature sensor (100°C warning and 150°C shutdown) for diagnostics and for protection of the power MOSFETs

dc42, 2017-05-29:
Yes.

notnyt, 2017-05-29:
That would explain why any value I entered for T ran the fan....

closed 2017-05-29 by notnyt

dc42, 2017-05-29:
The operands of the <= operator at lines 2107 and 2119 of Platform.cpp should be reversed. Will fix in next beta.