Duet3D/RepRapFirmware #108: Grid compensation not working (1.19beta2)

Alex9779, 2017-05-30:
Hi,
so I got another issue now as my machine is nearly finished.
I did a grid compensation run, all parameters seem fine, the heightmap is generated and looks ok. I have to do some manual leveling so but for now its ok.
But I thought after the run the leveling is active. Though no matter how I move a head, manually (panel or web), with the console (panel or web) or with a file the Z axis does not move.
I have deviations of about 0.3mm at max so I think I have to notice some movement. But there is any...
Did I miss something again?

Alex9779, 2017-05-31:
So I reverted to 1.18.1 and there without doing anything else I just move X around with the panel Z axis moves and compensates...

dc42, 2017-05-31:
Thanks for the report. Bug confirmed, and I am just testing the fix in 1.19beta3.

closed 2017-05-31 by dc42