Duet3D/RepRapFirmware #109: open load reported during printing

notnyt, 2017-06-01:
running 1.19beta3, seeing lots of these messages after updating from beta2 to beta3. 

1:26:18 AM
Error: Open load on drivers 2
1:26:13 AM
Error: Open load on drivers 2
1:26:09 AM
Error: Open load on drivers 3
1:26:02 AM
Error: Open load on drivers 2
1:25:57 AM
Error: Open load on drivers 2
1:25:49 AM
Error: Open load on drivers 2
1:25:14 AM
Error: Open load on drivers 2
1:25:09 AM
Error: Open load on drivers 2
1:25:05 AM
Error: Open load on drivers 2
1:24:59 AM
Error: Open load on drivers 2
1:24:36 AM
Error: Open load on drivers 2
1:24:31 AM
Error: Open load on drivers 2
1:24:26 AM
Error: Open load on drivers 2

dc42, 2017-06-01:
We were seeing this too in tests at one point. In theory it means you have a bad connection to the motor. We swapped over 2 of the motor connectors and it went away, so we concluded that we did indeed have a bad connection.

However, I don't rule out an issue with the driver reporting open load briefly when there isn't a problem. So if after checking your wiring you are still getting it, or you find there is a particular type of move that provokes it, let me know.

notnyt, 2017-06-01:
It's reporting opens for all of the drivers, but there's no open.  The motors are working normally.  I just wired this up last week and when I did, I tested it all out.

<details>
5:08:16 AM
Error: Open load on drivers 3
5:07:54 AM
Error: Open load on drivers 3
5:07:46 AM
Error: Open load on drivers 3
5:07:30 AM
Error: Open load on drivers 3
5:07:20 AM
Error: Open load on drivers 3
5:07:04 AM
Error: Open load on drivers 3
5:06:44 AM
Error: Open load on drivers 3
5:05:29 AM
Error: Open load on drivers 3
5:05:19 AM
Error: Open load on drivers 3
5:03:12 AM
Error: Open load on drivers 3
4:59:36 AM
Error: Open load on drivers 3
4:59:01 AM
Error: Open load on drivers 3
4:55:46 AM
Error: Open load on drivers 0
4:55:29 AM
Error: Open load on drivers 3
4:55:19 AM
Error: Open load on drivers 3
4:55:07 AM
Error: Open load on drivers 3
4:54:45 AM
Error: Open load on drivers 3
4:54:27 AM
Error: Open load on drivers 3
4:53:59 AM
Error: Open load on drivers 0 3
4:53:35 AM
Error: Open load on drivers 3
4:53:30 AM
Error: Open load on drivers 3
4:52:50 AM
Error: Open load on drivers 3
4:52:43 AM
Error: Open load on drivers 3
4:52:39 AM
Error: Open load on drivers 3
4:52:34 AM
Error: Open load on drivers 3
4:52:27 AM
Error: Open load on drivers 3
4:52:22 AM
Error: Open load on drivers 3
4:51:54 AM
Error: Open load on drivers 3
4:51:49 AM
Error: Open load on drivers 3
4:51:21 AM
Error: Open load on drivers 3
4:51:06 AM
Error: Open load on drivers 3
4:50:59 AM
Error: Open load on drivers 3
4:49:47 AM
Error: Open load on drivers 3
4:49:29 AM
[... 2589 lines cut ...]
1:26:13 AM
Error: Open load on drivers 2
1:26:09 AM
Error: Open load on drivers 3
1:26:02 AM
Error: Open load on drivers 2
1:25:57 AM
Error: Open load on drivers 2
1:25:49 AM
Error: Open load on drivers 2
1:25:14 AM
Error: Open load on drivers 2
1:25:09 AM
Error: Open load on drivers 2
1:25:05 AM
Error: Open load on drivers 2
1:24:59 AM
Error: Open load on drivers 2
1:24:36 AM
Error: Open load on drivers 2
1:24:31 AM
Error: Open load on drivers 2
1:24:26 AM
Error: Open load on drivers 2
</details>

notnyt, 2017-06-01:
actually, extruder is sounding funny when it's doing that...

dc42, 2017-06-01:
I have reproduced the problem, and released version 1.19beta4 with the open load detection disabled. Thanks for your report.

closed 2017-06-01 by dc42

notnyt, 2017-06-01:
0: No open load condition detected.
1: No chopper event has happened during the last period with constant coil polarity. Only a current above 1/16 of the maximum setting can clear this bit!
Hint: This bit is only a status indicator. The chip takes no other action when this bit is set. False indications may occur during fast motion and at standstill. Check this bit only during slow motion.

Interesting...

dc42, 2017-06-01:
Yes, we read that too. For now we have just disabled the open load detection, but we will re-visit it to see if we can correlate the open load flag with other parameters (e.g. speed and microstep position) to get a reliable indication.