Duet3D/RepRapFirmware #111: Extrusion issue after pause

notnyt, 2017-06-08:
Printing with extrusion in relative mode (M83 in gcode) and sending a pause from the paneldue

pause.g does not contain M82 or M83

Running G1 E-4 F3600 in pause.g causes a retraction which runs for quite some time, apparently going back to -4 as if it were in absolute mode

dc42, 2017-06-08:
Where did your pause.g file come from? I updated all the pause.g files in my github repo to include the M83 in August 2015.

notnyt, 2017-06-08:
made from scratch, works fine with M83 in there, just strange it changes mode when running pause.  If not worth looking at feel free to close

dc42, 2017-06-08:
Each gcode source has its own relative/absolute extrusion state. That state defaults to absolute extrusion, unless you include M83 in your config.g file. The state at the start of pause.g will be the state associated with whatever input source commanded the pause, i.e. web, PanelDue or USB.

I'd prefer to default the state to relative extrusion, but most slicers use absolute extrusion by default, so doing that would probably cause more users to request support.

closed 2017-06-08 by dc42

notnyt, 2017-06-08:
The issue was that even if the gcode source sets M83, then when pause is called, it goes back to absolute.

dc42, 2017-06-08:
Perhaps I didn't make myself clear. The SD card is a separate gcode source.

notnyt, 2017-06-08:
Oh, understood, thanks.

Eddiiie, 2017-06-08:
Ohhh is this why I cannot use Cura with DuetWifi/RRF/diamond head?

Only Slic3r works.   The extruders just grind grind grind never see so much filament come out at once.

What a mess.

dc42, 2017-06-08:
Sounds like you included M83 in your config.g file AND you forgot to include M82 in your Cura start gcode.