Duet3D/RepRapFirmware #112: Add Probing Speed Option to M201

eboston, 2017-06-18:
I just noticed the H option on the M906 to adjust the motor currents during probing.  Would it be possible to add a similar option for the acceleration settings?  When using the Precision Piezo probe, both the current and acceleration are recommended to be changed and with this option, there would not be a need to have a macro file to adjust the settings or to do it manually every time.

dc42, 2017-06-18:
It would be possible to add a new parameter to one of the M commands, but I'm not sure that it is needed. I think it's only the Z acceleration at the start of the probing move that might need to be reduced to avoid false triggering, and only on a delta printer because the Z acceleration on a Cartesian printer will normally be low anyway. Why not just reduce the Z acceleration in the M201 command in config.g? Unlike the X and Y accelerations, reducing the Z acceleration should have a negligible effect on print times, because Z movements are so small.

eboston, 2017-06-18:
But would it make sense to have the option for Delta printers?  I agree it probably isn't needed for Cartesian, but it sounds like it would for Delta.  Or is that already in place with the F option on the probe command?

dc42, 2017-06-18:
Just because a delta printer can do high Z acceleration doesn't mean that you need to set the Z acceleration the same as X and Y, or that there is any advantage in doing so.

closed 2020-10-20 by T3P3