Duet3D/RepRapFirmware #113: Abort G32 on a Z-Probe Error (also G29)

eboston, 2017-06-18:
Would it be possible to have the G32 command abort if there was a Z-Probe error?  If there is an error during the probing, it won't be able to do the calculations so no need to keep probing after an error.  It seems the G29 (mesh) is aborted on a Z-Probe error already, but not sure.

dc42, 2017-06-18:
Aborting execution of bed.g wouldn't suit users with deployable Z probes, because the command to retract the probe at the end of probing wouldn't be run. Perhaps it would be possible to skip any further G30 commands in the file.

eboston, 2017-06-18:
That would be fine.  When testing out a new Z-Probe, I was getting an error at the start, but still had to wait for all the other probes to occur before I could adjust anything.

T3P3, 2020-10-20:
This can be achieved with conditional gcode. 
https://duet3d.dozuki.com/Wiki/GCode_Meta_Commands

closed 2020-10-20 by T3P3