Duet3D/RepRapFirmware #114: DuetWifiFirmware1.19beta7

Eddiiie, 2017-06-23:
I applied this firmware through the WebUI.
Wifi no longer works.
Console on PanelDue says WiFi Module started Error: Failed to initiaise WiFi module code -10

How to fix?

dc42, 2017-06-23:
If you upgraded from 1.18.2 or earlier, see https://duet3d.com/wiki/DuetWiFiFirmware_1.19beta.

Eddiiie, 2017-06-23:
I pushed a button on the DuetWifi with a USB cable connected.
It said installing device driver for Bossic Port or something.
Then your reply came in.
I copied files to the SD card as mentioned.
Panel Due says "Connecting"

Computer connected via USB cable now says unrecognized device.

dammit

dc42, 2017-06-23:
So you pressed the Erase button, and now you need to re-install main firmware using SAM-BA or Bossac 1,.8. See the wiki.

closed 2017-06-23 by dc42

Eddiiie, 2017-06-23:
Really?  Thanks for the help.

Eddiiie, 2017-06-23:
SAM-BA has a list of a bunch of boards, Wiki does not help with this.

Eddiiie, 2017-06-23:
ok I got it.. Fallback 3 -   https://duet3d.com/wiki/Updating_main_firmware