Duet3D/RepRapFirmware #118: Z tool offset only considered when positiv (1.19beta6)

Alex9779, 2017-06-27:
I wanted to calibrate my tool offsets, X and Y are working fine but my second tool is sllightly higher than the the first one.
IMHO I need a negative Z offset but this is not working at least for Z values lower than the offset. For exmaple Z=0.
Setting a negative offset to the tool and switching to it does not move the Z axis, the display just shows the postive number of the negative offset.
Setting a positive offset works, switching to the tool moves Z.
I assume this has something to do with the set limits because the base Z limit is 0 and applying that offset would result in a negative Z value.
Any ideas other than remounting the extruder?

dc42, 2017-06-27:
The axis limits are applied after tool offsets are applied. This obviously makes sense for the X and Y offsets, but possibly not for the Z offset. Here are 2 workarounds:

(1) Define the higher tool as having Z offset 0, then the lower one will have a positive Z offset. Then redefine the homing switch offset and/or Z probe trigger height to get Z=0 correct.

(2) Use the M208 S1 command to allow movement a little below Z=0, for example M208 S1 Z-1 will allow movement down to Z=-1mm.

Alex9779, 2017-06-27:
> but possibly not for the Z offset
do you mean you think about changing something to allow this?

I think I'l use (2), thanks, seems not so invasive, just a risk when moving manually...

Alex9779, 2017-06-28:
Ok I just tried (2) with 1.19beta6.
I set the Z minima to -0.2, the offset of the second I have is 0.17mm so this works now...

closed 2020-04-22 by Alex9779