Duet3D/RepRapFirmware #120: Program "arm-none-eabi-g++" not found in PATH

Gigga-Wc, 2017-07-04:
Hello!
I need to do some changes in the corexy kinematics becouse i'm design a printer with a unusual way of movement.
So i downloaded all files and go step by step through the buildinstruktions.
First i struggle with eclipse neon 2, this Version is'nt downloadable anymore.
So i go with Oxygen first. But witch one i dont know? Java or c++ or something else?
I gave the two a Chance, always with the same results: Program "arm-none-eabi-g++" not found in PATH
Next was neon 3... same issue.
Then i tried the 32bit Version of all those eclipse's.
Im also deleted the Firmware Folder every time i try a new eclipse and pushed a fresh copy of the fileset into it.

So please help!
[image]

larsarv, 2017-07-04:
Hi Gigga-Wc!
arm-none-eabi-g++ is not part of eclipse. You can get it from https://launchpad.net/gcc-arm-embedded/+download (need the corrext version) or from Arduino version 1.5.8 as decribed in the BuildInstructions.md...

Gigga-Wc, 2017-07-04:
Hello larsarv!
I downloaded the Arduino 1.5.8 IDE and used it in eclipse as you can see in my screenshot. I hilighted in yello.
Do i need a other, newer version? There are thousands in your link.
Thanks, Christian

larsarv, 2017-07-04:
Hi,

I used gcc-arm-none-eabi-4_8-2014q3 for mac! Latest version do not work.
Eclipse version I use is eclipse-cpp-neon-3-macosx-cocoa-x86_64 and I only changed the "cross settings" to point to it as you show in the picture...

Gigga-Wc, 2017-07-04:
Ok, Thanks!
I give this 4_8-2014q3 Version a try.

larsarv, 2017-07-04:
Btw, I have the same two errors and can build with no problem anyway...

larsarv, 2017-07-04:
To get rid of the error: http://www.eclipse.org/forums/index.php/t/697857/

Gigga-Wc, 2017-07-04:
I also try to build it, takes  112milliseconds with no error Pops up.
But i dont get out a file or i just dont know where this file is.
If i push the debug button a Msg Box with  "Program "arm-none-eabi-g++" not found in PATH" Pops up

larsarv, 2017-07-04:
Try building the RepRapFirmware project, the result is DuetWiFi/DuetWiFiFirmware.bin (if you build for duetwifi). You cant run or debug it on your computer me think, need to upload it to the duet.

Gigga-Wc, 2017-07-04:
So there are no more Errors with  this 4_8-2014q3 Version Thanks!
But i dont get out a file anyway.
I use the Open projekt from file System... Option to Import, and the hammer button to build.
no Errors.... no file???

Gigga-Wc, 2017-07-04:
Also tryed the build all. And yes it's for duetwifi

Gigga-Wc, 2017-07-05:
**Problem solved**

After i searched a hole in the worldwideweb, and downloaded the half of it, i found a solution.
reprapprochina-liang has discovered that the 
gnu make builder dosent work and used the CDT Internal builder instead.
http://forums.reprap.org/read.php?416,718274

This should be pointet out in the buildinstruktions.md file, also that you need the c/c++ eclipse.
Also eclipse Oxygen works just fine.

closed 2017-07-05 by Gigga-Wc