Duet3D/RepRapFirmware #124: [Proposal] New GCode to upload binary files

cangelis, 2017-07-14:
Currently, I develop an application on a linux board (with python -similar to OctoPi-) to manage duet board. What I'd like to do is to upload the configurations and .bin files to Duet from linux board so I can automatize the firmware update process and linux board can easily download and update the Duet's firmware.

There are FTP, Telnet and HTTP options for uploading but these options require a network connection between my linux board and Duet. This options are not valid if the linux board's network hardwares are already in use by the user.

So what I propose is to make a new GCode (or update an existing one) to upload binary files. Currently, there is `M560` for uploading HTML files that is similar to what I need.

So my first attempt was to modify `M560` to upload binary files to `sys` folder rather than `www`.
That worked ok for configuration files but not for the firmware's bin file. 

Because `DuetWiFiFirmware.bin` file already contains `EOF_STRING` (``) in it, so uploading this file is not possible because uploading always finishes before the end of file while reading pointer encounters the `` in the file.

My solution was to provide the file size to the GCode so program can know when to finish reading. I already implemented a proof of concept (-which breaks `M28`-).  Here it is: https://github.com/cangelis/RepRapFirmware/commit/6c740a2dd6c160176d47bc8ae50194e7ee244405

I use `M559 S<file size> PDuetFirmware.bin` to upload the file.

If this idea makes sense, We can discuss about the new GCode or modifying the existing one.

dc42, 2017-07-15:
Thanks for your proposal. I think this deserves wider discussion. Here are some points to consider:

- Should we really be sending binary data? Or should we send e.g. base64-encoded data, perhaps with a checksum at the end? This would allow us to use a unique end-of-data string that cannot occur in the encoded data. It would slow down transmission slightly because we would be encoding 6 bits per character instead of 8.

- When would the binary data be deemed to start? Bearing in mind that the line that contains the file upload command could be terminated by \n, \r or \r\n depending on the operating system? This creates a potential ambiguity.

- Instead of passing the length of the data as a parameter in the command, we could pass the end-of-data string as a parameter. This would be a minimal modification to make, especially now that RRF now supports quoted string parameters. If you are looking to implement something quickly, then I suggest we add this option to the existing command.

chrishamm, 2017-07-16:
I like the idea of adding support for binary transmissions to the G-Code interface. In principle your proposal about using the 'S' parameter for the file size sounds good to me, but I have a few notes on your implementation:

1) Don't use floats for file sizes, use the FilePosition type instead as we're operating on FAT. Also instead of seekSize it may be sufficient to use fileBeingWritten->Length().
2) If we go with the 'S' parameter, we should support it in M560 as well. Compressed DWC files are in the GZ format which would need binary encoding. Besides there is still M28 to upload regular G-code files. Should we add support for binary uploads to this M-code as well? This way we would keep the comments, which could be utilised by the PrintMonitor class.
3) If the USB cable is unplugged during an upload, the firmware should automatically close and delete the file being uploaded. This isn't implemented yet and could lead to problems.

Here my thoughts on David's comment:

- I wouldn't want to upload base64-encoded data because it increases the amount of transmitted data and because it requres the whole data to be converted first. Using a file size instead sounds better to me.
- Checksums are a nice idea but I wouldn't want to include them in the uploaded data. How about adding a new 'C' parameter to the M-codes to specify an optional CRC32 checksum? I guess it would be a good idea to add a CRC library in the future anyway if we want to send files through the SAM-BA interface to the ATSAM4S.
- AFAIK no OS uses \r to terminate the line, so I suggest to always start file uploads via (M28?)/M569/M560 when a \n character is received.
- In general I have no problem with specifying the binary file size - this is already done by the scanner module and it works perfectly well. However we need to retain compatibility with the old-fashioned upload protocol (i.e. M29/EOF string) because that is used by Pronterface AFAIR. It should be sufficient to check that only if no file size is specified.

dc42, 2017-07-16:
Chris, thank for your comments. My comments

1. With Base64 encoding, it isn't necessary to convert the whole file in one go. You can read the file in chunks and use the Java library to convert each chunk, see https://docs.oracle.com/javase/8/docs/api/java/util/Base64.Encoder.html. But you do get an overhead of 33% in the size of the transmitted data.

2. Mac OS versions prior to OSX use \r alone as the line ending. Perhaps we don't need to support those.

3. I like the idea of a separate CRC parameter.

WZ9V, 2017-07-16:
Rather than base 64 why not use the old BBS Z-Modem protocol.  It has the ability to restart, handles binaries, etc.

cangelis, 2017-07-17:
- I think base64 encoding & decoding is another cumbersome for the firmware and the client in terms speed, efficiency and simplicity.
- Sending size to the receiver is already used by modern protocols including HTTP and FTP. This makes file transfer easy and more reliable & consistent comparing to sending eof string. 
- Checksums are always a good idea.
- What about defining a new parameter `F` to `M560` to set the folder rather than defining different gcodes (`M559`, `M560`) for each folder? F=0 for sys, F=1 for www, F=2 for macros etc or just F/sys maybe?

dc42, 2017-07-17:
The folder should be passed as part of the filename, that's how all the other file-related gcodes work.

Here is a proposed specification:

>>
M559 and M560 shall perform identically except that the default folder for M559 is /sys and the default folder for M560 is /www. The default folder is used if the filename provided does not include an absolute path, i.e. one that starts with / or n:/ for some integer n.

The parameters to these commands shall be:

Pfilename or P"filename" - compulsory. The quotes must be used if the filename contains certain characters such as ';' and '(' and ' ' or if any other parameters are provided.
Snnn - file size in bytes, optional. If not provided then the file ends when the EOF string is found.
Cnnn - CRC-32 of the file, optional. If the CRC of the received data doesn't match the file, then the received data is not stored and an error is reported.

The file data begins after the \r\n or \n that terminates the line containing the command.
<<

Comments?

chrishamm, 2017-07-17:
Sounds all good to me but what about M28? It would make sense to enhance it too.

dc42, 2017-07-17:
M28 takes a filename without a parameter letter, so there is no option to add more parameters unless we subvert the existing definition. I suggest we leave it alone.

cangelis, 2017-07-17:
Just opened a PR we can continue on it. #125

closed 2017-07-17 by cangelis