Duet3D/RepRapFirmware #126: Pressure advance stuttering with small moves simplify3d 4.0

notnyt, 2017-07-27:
I'm slicing with simplify3d 4.0 which ends up with some small moves.  It's easy to see on the inner layer of the skirt here. Pressure advance causes some strange stuttering.

The sliced file is attached, settings are as follows.  Video available displaying the issue at the link below.

https://www.youtube.com/watch?v=BEV0s_uiXXk
````
M203 X7200 Y7200 Z1200 E3600        ; Maximum speeds (mm/min)
M201 X1500 Y1500 Z400 E1800         ; Accelerations (mm/s^2)
M566 X600 Y600 Z24 E1800            ; Maximum jerk speeds mm/minute
M572 D0 S0.07
````

sample of moves
````

G1 X208.663 Y216.649 E0.0002
G1 X208.285 Y216.781 E0.0166
G1 X208.281 Y216.782 E0.0002
G1 X206.504 Y217.328 E0.0771
G1 X206.500 Y217.329 E0.0002
G1 X206.134 Y217.427 E0.0157
G1 X206.130 Y217.428 E0.0002
G1 X205.077 Y217.666 E0.0448
G1 X205.074 Y217.667 E0.0001
G1 X204.810 Y217.719 E0.0112
G1 X204.807 Y217.720 E0.0001
G1 X203.674 Y217.914 E0.0477
G1 X203.671 Y217.914 E0.0001
G1 X203.331 Y217.961 E0.0142
G1 X203.328 Y217.961 E0.0001
````
[3DBenchy.zip](https://github.com/dc42/RepRapFirmware/files/1178730/3DBenchy.zip)

notnyt, 2017-08-10:
This is from 1.19beta11

dc42, 2017-08-10:
This is due to a bug in S3D. See my response on the duet3d forum.

closed 2017-08-10 by dc42