Duet3D/RepRapFirmware #129: Message boxes don't pass axis correctly

kitsuneaerin, 2017-08-25:
When using M291 to pass a message with any of X, Y, or Z axis specified, the sent alert is always 1; This is because platform::SendAlert only takes bools, while it's passed an axisControls. I'm working on a pull request now to handle this, since there's only three references to SendAlert, one of which passes false, and the other to do with manual bed probing, which should (I assume) give just Z control. I'll reference this issue when done.

dc42, 2017-08-25:
Thanks for the report. No need for the PR, I was about to test a provisional release anyway, and I'll include a fix for this problem.

dc42, 2017-08-25:
I've committed my fix. I think your change to  GCodes.cpp wasn't quite right. Please check.

I still need to run some tests on 1.19+3 before inviting users to try it.

kitsuneaerin, 2017-08-25:
I noticed that. Didn't quite click that that was a bitmap for some reason. Getting a bit rusty! Your edit is far better

closed 2017-08-27 by kitsuneaerin