Duet3D/RepRapFirmware #131: Issue with new M104 standby temperature feature

dotorg, 2017-09-05:
In trying to get the tool and temperature behavior in RepRapFirmware on my Duet working with the behaviors the various slicers all seem to presume (specifically, M104/M109 used to change temperatures, not G10, and not having the concept of standby temperatures), I've struggled to get my dual-extrusion working with slicers properly, especially with smart temperature control. That said, if the slicer is changing first-layer temperatures for a filament, there's problems, too. 

The updated firmware that sets the standby temperature for inactive tools when M104 is called is introducing a pretty substantial behavioral inconsistency that has made the situation worse. In my testing, when both Cura and Slic3r generate the G-code to change temperatures after the first layer, only the inactive tool is being properly updated. The active tool maintains its previously set standby temperature, adding a significant time on tool changes on each layer as temperatures rise or drop from the first layer temperature (set on standby) to the correct temperature.

The prior behavior wasn't great, but at least was consistent. 

I think the system should replicate the active and standby temperatures on *any* tools set on the "legacy" temperature (M104 *and* M109) g-codes, regardless of being active or inactive. Any call to either of them should set both to that value to preserve compatibility with existing slicer logic. Slicers that have been updated to use G10 can manage the standby temperatures as appropriate.

larsarv, 2017-09-05:
> I think the system should replicate the active and standby temperatures on any tools set on the "legacy" temperature (M104 and M109) g-codes, regardless of being active or inactive.

I have yet to test the new firmware with the modified M104 but what dotorg is suggesting sounds like the exact modification (on line added) I apply to every version of the firmware I install.
I did a request for this behaviour on the firmware wish list some time ago.

dotorg, 2017-09-18:
Any chance you could post what you change? This is such a huge issue for essentially every single slicer, it makes it difficult, if not impossible, to do multi-hotend prints with RepRapFirmware... There's no way other than manually monitoring it at the start of the print to ensure everything consistently gets set correctly, and its impossible to vary temperatures correctly mid-print. 

It'll save me having to dig through the code to find where to patch it, but its clear Im going to have to just do so.

larsarv, 2017-09-19:
In GCodes.cpp (version 1.19) I added "standby[h] = temperature;" to GCodes::SetToolHeaters. 

```
void GCodes::SetToolHeaters(Tool *tool, float temperature)
{
	if (tool == nullptr)
	{
		platform.Message(GENERIC_MESSAGE, "Setting temperature: no tool selected.\n");
		return;
	}

	float standby[Heaters];
	float active[Heaters];
	tool->GetVariables(standby, active);
	for (size_t h = 0; h < tool->HeaterCount(); h++)
	{
		active[h] = temperature;
		standby[h] = temperature;
	}
	tool->SetVariables(standby, active);
}

```

T3P3, 2020-10-20:
If the current handling of M104 etc is still an issue please discuss this on the forum (forum.duet3d.com)

closed 2020-10-20 by T3P3