Duet3D/RepRapFirmware #133: Bed leveling with independent Z motors

Kriechi, 2017-09-09:
Wiki: https://duet3d.com/wiki/Bed_levelling_using_multiple_independent_Z_motors

I'm running DuetWifi with FW 1.19.2 and a CoreXY setup.
I want to do bed leveling with G32 with two independent Z motors, and two leadscrews on the center-left & center-right of my bed.

My config:
```
M584 X0 Y1 Z2:4                                 ; Drive mapping: 0=A, 1=B, 2 and 4 are individual Z steppers
M671 X-52:296 Y142:142 S15                      ; Define positions of Z leadscrews
```

My `bed.g`:
```
M561                        ; clear any existing bed transform
G28                         ; home
G30 P0 X10 Y142 Z-99999
G30 P1 X230 Y142 Z-99999 S2
G28 Z
```
This gives me the error that my points are not clockwise oriented. I suppose there is some math trickery involved, so I change Y to be 1mm less:

```
M561                        ; clear any existing bed transform
G28                         ; home
G30 P0 X10 Y142 Z-99999
G30 P1 X230 Y141 Z-99999 S2
G28 Z
```
Ok, now I get this error: `Bed calibration error: 2 points provided but only 3, 4 and 5 supported`
Which is weird, because why would I want to probe 3 points on my 2-leadscrew setup?
As far as I can tell, there is no code for a 2-point calibration: https://github.com/dc42/RepRapFirmware/blob/cfbe3562cc52de52bed88f7360f36b64a8049571/src/Movement/BedProbing/RandomProbePointSet.cpp#L89

Just for the sake of completeness, I added a third point (back-center). Now I finally get a bed map and some offset information. But still not auto-correction.

I think I want to somehow trigger this (https://github.com/dc42/RepRapFirmware/blob/8dda9ff9fa096a9093d0c85c276c43f140267e0f/src/Movement/Move.cpp#L846), but fail for unknown reasons.

dc42, 2017-09-10:
Please ask support RRF questions on the forum at duet3d.com.

closed 2017-09-10 by dc42

Kriechi, 2017-09-10:
I believe this to be a bug, since I couldn't trace the code path to where it should go...
Anyway, I moved it to the forum: https://www.duet3d.com/forum/thread.php?id=2599