Duet3D/RepRapFirmware #134: Trinamic coolStep

KyleMaas, 2017-10-02:
In the TMC2660 register configuration, coolStep is disabled for the following reason:

// disable coolStep, we already do this in the main firmware

This is not entirely true.  Yes, the firmware lowers the motor current if a motor is idle for long enough, but it doesn't react nearly as quickly as coolStep would, particularly movement.  On a long move, coolStep has the potential to drastically reduce the amount of noise emitted between the acceleration and deceleration phases.  I would love to see this feature enabled, or at least made to be a configurable option.  Reading through the TMC2660 datasheet, I would think a reasonable starting point would be:

TMC_SMARTEN_SEIMIN_QTR | TMC_SMARTEN_SEDN_4 | ((0x03 & TMC_SMARTEN_SEMAX_MASK) << TMC_SMARTEN_SEMAX_SHIFT) | TMC_SMARTEN_SEUP_8 | ((0x01 & TMC_SMARTEN_SEMIN_MASK) << TMC_SMARTEN_SEMIN_SHIFT)

That way, as I understand it, it would slowly lower the current if it wasn't needed but would quickly increase it if it was needed.

For easy reference, here is a link to the datasheet:

https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2660_datasheet.pdf

Thank you for your consideration.

dc42, 2017-10-03:
Thanks for your input. I have re-read the coolstep section of the datasheet.

The only advantage that I can see to using coolstep is to reduce power consumption and heat generation in the motors during fast movement. Motor noise is already very low using the x256 interpolation mode, so I doubt that the noise reduction would be significant. During printing moves, reducing motor current would cause reduced microstep accuracy, with a likely reduction in print quality.

It is clear that the coolstep thresholds need to be tuned to the motors in use, so it's not simply a matter of enabling it.

I have been asked to make the stallguard configuration parameters accessible in a future version of the firmware, so I will consider making the coolstep register accessible at the same time. The chopper control register is already accessible.

KyleMaas, 2017-10-03:
My machine is a fairly large CoreXY configuration, and as such, it has a
decent amount of mass on the Y axis for the gantry.  Even with x256
interpolation, the machine is still quite noisy.  I can turn down my
motor power, but then I can't run the machine nearly as fast or get the
accuracy I'm current able to run with - it really needs the power to be
able to push things around.  But once that mass is moving, it's quite
easy to keep it moving, so I could see a huge benefit to adding coolStep
for noise reduction.  Even if it was just enabled for the long
non-printing moves where microstepping accuracy isn't as important and
disabled during printing moves, it would be a huge benefit.  At any
rate, even if it's not enabled as a default, it would be great if I
could configure it without having to compile the firmware to do so.

dc42, 2017-10-31:
This is resolved in firmware 1.20beta2.

closed 2017-10-31 by dc42

KyleMaas, 2017-10-31:
Very much looking forward to trying it.  Thank you!

KyleMaas, 2018-02-02:
I just wanted to say: thank you for implementing this!  I've got it mostly tuned at this point, and it makes a huge difference in how noisy my printer is.  So far I seem to get my best results with this (back-translating this to constants, since I mostly did this using the bit fields in the datasheet):

TMC_SMARTEN_SEIMIN_QTR | TMC_SMARTEN_SEDN_1 | (2 << TMC_SMARTEN_SEMAX_SHIFT) | TMC_SMARTEN_SEUP_8 | (5 << TMC_SMARTEN_SEMIN_SHIFT)

(This comes out to a decimal value of 57957, if you want to try it.)

Prints are very smooth, and I keep wondering if the printer's still printing, because I'm not used to not being able to hear it.  So, again, thank you!