Duet3D/RepRapFirmware #138: layer count erroneous on Web Montior

gratefulfrog, 2017-10-21:
Running version 1.19, today I noticed this display on the upper right of the monitor:

`Layer: 98 of 82, Filament Usage: 2912.7mm of 4786.7mm (1874.0mm remaining)`

At that point Z was at approx 10.2 and the object total height was 12.45 mm with a Layer Height of 0.3 mm / 0.15 mm.

This issue is new in version 1.19.

dc42, 2017-10-21:
Please provide the gcode file.

gratefulfrog, 2017-10-21:
Here it is !

[drain_7_20170617_no_homing_no_bed_compensation.gcode.zip](https://github.com/dc42/RepRapFirmware/files/1404041/drain_7_20170617_no_homing_no_bed_compensation.gcode.zip)

dc42, 2017-10-21:
Looks like a slicer bug, or you are printing several separate objects with different layers heights, or you are printing infill with a different layer height. The Z coordinates in the gcode for each layer are:

0.3, 0.3, 0.4, 0.45, 0.6, 0.7, 0.75, 0.9, 1.0, 1.05, 1.2, 1.3, 1.35...

So the number of different heights (which is how RRF counts layers) is much more than 82. I doubt that this behaviour is new in 1.19.

gratefulfrog, 2017-10-21:
I had never seen this error before using version 1.18

The layer heights are correct, in my opinion.

dc42, 2017-10-21:
If you believe the layer height are correct even though they are not evenly spaced after the first layer, you need to explain why you think that. After the first layer the layer height appears to cycle between 0.1, 0.05 and 0.15mm.

gratefulfrog, 2017-10-21:
I am using slic3r 1.2.9. The part has not been re-sliced since it was made.

The reason for the layer spacing is because of the way the supports are printed: 0.1 + 0.05 = 0.15.

I have printed this part many times on the previous 1.18 web interface and never noticed an issue with the layer count, but I cannot be 100% sure of that. However, I monitor all my prints via the monitor and had there be that kind of very apparent error, I am nearly positive I would have noticed.

Also, the finish time prediction fails for the layer prediction.

In any event, since it seems relatively easy for a person to count the layers, it should be easy for the program, should it not?

I only reported this to help, not to cause any grief.

Cheers,
Bob

dc42, 2017-10-21:
I've made a change in the forthcoming 1.20beta2 firmware that I hope will resolve this. Please try it when I release it, probably early next week.

gratefulfrog, 2017-10-22:
Great! I will ! Let me know if you think of it!

dc42, 2017-10-31:
Believed resolved in 1.20beta2.

closed 2017-10-31 by dc42