Duet3D/RepRapFirmware #140: HTTP query parameters assumed position?

Kriechi, 2017-10-21:
I noticed that some parts in the `Webserver::HttpInterpreter` assume a fixed position / ordering of HTTP query parameters.

For example: https://github.com/dc42/RepRapFirmware/blob/f437e684e7dd4c991b29ffb2b93db7595c5da05e/src/Duet/Webserver.cpp#L908, where `time` **must** be the second parameter.
https://github.com/dc42/RepRapFirmware/blob/f437e684e7dd4c991b29ffb2b93db7595c5da05e/src/Duet/Webserver.cpp#L1530, same here with `name`.

I'm not sure if the ordering / position is somewhat defined in an RFC, but IMHO the position in which a key-value pair occurs in URL-encoded query should not matter.

And since some HTTP clients just use random order (e.g., when serializing hashmap/dicts) this can cause headaches.

dc42, 2017-10-21:
Will be fixed in 1.20beta2. Thanks for your report.

dc42, 2017-10-31:
Resolved.

closed 2017-10-31 by dc42