Duet3D/RepRapFirmware #143: Cancel print automatically if bed probing fails

Alfaa123, 2017-11-09:
We have been having issues with our IR probes disconnecting at the connector for various reasons. It seems that when the probe gets connected, the signal level floats and either triggers all the time or doesnt trigger at all, randomly.

We get an error that says "Z probe triggered at start of probing move" and the bed compensation is cancelled. The print, however, continues even with the bed compensation and the z height potentially wrong, giving is disastrous results.

Is there a way to make the print cancel if it encounters an error, or is this functionality not implemented yet?

dc42, 2017-12-06:
I'm reopening this because it's something to consider adding.

Alfaa123, 2017-12-06:
Ah ok, fair enough.

It didn't receive any attention when I first posted it, so I kinda assumed it was either already implemented, or I was missing something.

I did see awhile back that forcing the IR sensor to operate in digital mode had the effect of always triggering if the cable became disconnected, instead of never triggering if disconnected in analog mode, which seems preferable.

Maybe future versions of the probe can have their analog signalling inverted?

Anyway, our problem was mostly solved by removing the connector entirely and soldering directly to the board.

ragd011, 2019-03-24:
If not appropriate please let me know and I can create a separate FR.  But along the same lines, once conditional is a thing, would it be possible to specify a max deviation on a delta?  If it can't achieve less than that value it prints a message and doesn't start the job.  I've had a few prints that come out skewed when it has a calibration problem.  Most likely because I had a little cold PLA on the nozzle and the FSR tripped early.  Again, sorry if this is in the wrong place.  Thanks, Rob.

T3P3, 2020-10-20:
This can now be handled with conditional gcode:
https://duet3d.dozuki.com/Wiki/GCode_Meta_Commands

closed 2020-10-20 by T3P3