Duet3D/RepRapFirmware #144: Usage for the M305 S"name"

timmit99, 2017-11-10:
Was/is there a planned use for the "S" parameter in M305? I would love to have these names be displayed on the web interface instead of the generic "Heater 1" name. I was curious if there was already a use for them that i didn't discover yet or if it is planned for the future.

dc42, 2017-11-10:
If you create a virtual heater and use the S parameter to give it a name, it will be displayed in DWC in the "Extra" temperatures tab.

timmit99, 2017-11-10:
What about non-virtual heaters? 
My current command is M305 P2 S"Left Extruder" X205 H0 L0 for an FDM extruder using a PT100 on channel 205. It would be nice if the web interface could pull the heater names into the web interface.

dc42, 2017-11-11:
I suggest you request that as an enhancement to Duet Web Control, in either the github issues list for that project, or at https://www.duet3d.com/forum/viewforum.php?id=11. It will need changes to this firmware too, to send the names across.

timmit99, 2017-11-11:
OK, I have already made a request on the Duet Web Control page. I'll see what the response is and reply here.

closed 2018-08-27 by dc42