Duet3D/RepRapFirmware #145: DHT22 Read error

devmsp, 2017-11-17:
Hello, need a little help setting up DHT22 sensor for my chamber heater. I'm using DuetWifi board with Duex5 connected. I'm using the Temp Daughterboard connector on Duex5 for connecting DHT22 sensor. I have connected DATA wire to CS5 (Pin 3) , VCC to 3.3v (Pin 8) and GND to GND (Pin 2). I have connected a 10K resister between VCC and Data line of DHT22 near the sensor.

Following is my firmware versions right now.

Firmware Name:	RepRapFirmware for Duet WiFi
Firmware Electronics:	Duet WiFi 1.0 + DueX5
Firmware Version:	1.20beta7 (2017-11-11)
WiFi Server Version:	1.20b8
Web Interface Version:	1.19.3

In Config.g I have added the following code.

M141 H3  ; heater 3 is the chamber heater
M305 P3 X400 T"22"
M301 H3 P-1

but for some reason the Chamber Heater temperature in the Web Interface is always showing 2000 C. I have tested the sensor with Arduino and it's working correctly there. I will greatly appreciate any help you can offer.

Thanks & Regards
Dev

chrishamm, 2017-11-17:
Do not put the "22" in quotes. Instead of

M305 P3 X400 T"22"

you should be using

M305 P3 X400 T22

closed 2017-11-17 by chrishamm

devmsp, 2017-11-17:
Thanks a lot for your quick response. I tried removing the quotes but that didn't solved the issue. It's still showing 2000 C in chamber heater.

I haven't yet connected a heater yet, could that be a problem?

chrishamm, 2017-11-17:
I know the underlying code works because I know someone who is using the same module, so I figure the problem is somewhere in your wiring. Try replacing the 10k resistor with a 5.1k resistor - this page may help too: http://hivetool.org/w/index.php?title=Sensors:_DHT22

I have a DHT11 break-out board working on my own printer as well.

devmsp, 2017-11-17:
Thanks a lot, I will try changing the resistor and check if that solves the issue.

devmsp, 2017-11-17:
I got it to work finally but somehow the sensor is only working with pin CS6, I'm using the wiring diagram here https://duet3d.com/wiki/File:Duex5_connectionsv0.8_d1.1.png for reference. It's weird cause the instructions on Duet Wiki said it must be CS5 pin. 

It's working and showing accurate temperature now though with CS6 pin and following code.

M141 H3  ; heater 3 is the chamber heater
M305 P3 X400 T22
M301 H3 P-1

Thanks for your support and is there any plans to show Humidity value anytime soon?

chrishamm, 2017-11-19:
You're right - it ought to be pin CS6. I have changed that on Wiki page.

The humidity value is already supported, use thermistor channel 401 for it. If using the latest DWC beta 4 from my GitHub repository, you can specify the unit as well like this:

M305 P102 X401 S"DHT humidity [%]"

devmsp, 2017-11-20:
Thanks a lot Chrishamm, for all your guidance.