Duet3D/RepRapFirmware #146: Looking for a way to add a Z-probe failsafe

C1rcu1704444, 2017-11-27:
Looking for a way to detect if the Z-probe is plugged in or not. Have had an issue with the Z-Probe coming unplugged.  Recently had a crash when the probe came lose. This should not regularly be a problem since the connection should be solid but with this system we cannot afford to risk it and we are constantly moving things around and plugging and unplugging pretty much everything.

Thanks

timmit99, 2017-11-27:
I think that having an M583 in the HomeZ and HomeAll macros would solve this if you had a return voltage from the Vcc on the connector. If you added a wire connected to the board which would return Vcc when it is plugged in (via  a jumper on the board or something) it would only return a voltage when the connector is attached. This could then be tied to a pin which would be able to be used with the M583 "Wait for pin" command; waiting for the pin to be high before homing the Z.

@dc42 any ideas on when M583 will be implemented?

dc42, 2017-11-27:
The Z probe is already failsafe if you use a probe with an active high output (as most types of Z probe are, except for NPN-output inductive or capacitive probes), and select type 5 in the M558 command. If no probe is connected, it will report as triggered.

The same is true if you use type 4 or type 6.

C1rcu1704444, 2017-11-27:
Thank you for the response.
We are using Mini Differential IR height sensing board. To my understanding that outputs an analog signal when connected to the duet, and that would mean we need to use type 1 on M558?  Is there a failsafe built in for that as well?

dc42, 2017-11-27:
If you select probe type 5 in M558 instead of type 1, then the IR sensor will start up in digital mode. A possible disadvantage is that in analog mode, the probing speed is reduced when getting close to the trigger point; but this doesn't happen in digital mode.

C1rcu1704444, 2017-11-28:
Thank you for the help!

closed 2020-10-20 by T3P3