Duet3D/RepRapFirmware #147: Turn off tool switching behavior of M109 Tnnn

jpwheless, 2017-12-02:
A couple releases ago the behavior of M109 and M104 was changed so that they update temperatures immediately without having to use a tool switch command, which is perfect, but now M109 Tnnn also switches tools by itself.  Both Cura and Slic3r expect the old behavior of M109 where it has no effect on which tool is active.

Cura's default start code (in RRF mode) is the following:

`T0
M190 S60
M104 S230
M104 T1 S245
M109 S230
M109 T1 S245

<start using T0 here>`

I have to fix the generated gcode each time by inserting a T0 right after the M109 T1 command, or else it starts out using the wrong extruder.  Cura already has a few issues open regarding the start code temperatures for dual extrusion, but it doesn't sound like they're going to be fixed anytime soon.

Can M109's behavior be switched back to not changing the active tool?

dc42, 2017-12-03:
It looks like we can't win! We had a lot of users failing to print because no tool was selected. Most slicers don't put a T0 command in the starting GCode for single-material prints. Worse, most slicers put the temperature-setting commands *before* the user's own start GCode, so even if the user includes T0 in the start GCode, it's too late. To avoid this, I had M109 select the tool, on the grounds that a command to set temperature and wait doesn't make sense if no tool is selected.

I'm reluctant to remove this feature, because if I do then we'll get lots of users having this problem again. Maybe there is a compromise that will satisfy most users, such as:

A. If M109 is issued when no tool is selected, select a tool, otherwise don't.
B. As A but only do this is if exactly one tool is configured.

Your thoughts?

jpwheless, 2017-12-04:
That workaround makes sense.  I hadn't known why it had been changed.

I think your option A would work.  I'd make the behavior of M104 the same too, because right now if a tool is off then it won't respond to M104.

Also, right now it looks like when M109 or M104 is used on an active tool, it only sets the active temperature.  If M109 or M104 is used on a tool that's off or in standby, it sets both active and standby temperatures (and M104 doesn't turn a tool that's off, on).  Since slicers that don't use G10 have no concept of firmware-controlled active or standby temperatures, I think M109 and M104 should set both temperatures all the time.

(M104 OR M109) Snnn
- If no tools are active, set Tool 0 to active
- Set active tool's active and standby temperatures to Snnn

(M104 OR M109) Tnnn Snnn
- If no tools are active, set Tnnn to active
- If another tool is active but Tnnn is off, set Tnnn to standby
- Set Tnnn's active and standby temperatures to Snnn

What do you think of that behavior?

jpwheless, 2017-12-14:
Issue was fixed in V1.20 RC1.  Thanks!

closed 2017-12-14 by jpwheless