Duet3D/RepRapFirmware #149: Cannot Print Files With Parenthesis

Alfaa123, 2017-12-22:
Any files with a parenthesis in it's filename will fail to print with a "File Not Found" error.

Removing the parenthesis solves the issue. Problem occured after update allowing parenthesis in gcode commands, so I believe the parenthesis is acting as an escape character and the inputs were not sanitized.

dc42, 2017-12-23:
In standard GCode, anything in parentheses is a comment, so it is ignored. In firmware 1.20, in any command that take a filename or other string parameter, you can enclose that parameter in double quotation marks. This allows you use use spaces, parentheses, semicolons etc. in filenames.

closed 2017-12-23 by dc42

Alfaa123, 2017-12-23:
Oh, of course.

But when printing from a PanelDue or the web interface, it simply fails with no explanation.

Obviously, this isn't a super important issue, but I simply figured I'd try and contribute where I can.

dc42, 2018-01-10:
It's fixed in the current PanelDue beta firmware,