Duet3D/RepRapFirmware #151: Support ideamaker's info comments

andrasbiro, 2018-01-02:
I started playing with ideamaker and realized RRF doesn't support its info comments, therefor the print time estimations are not too good.
Example gcode:
https://www.dropbox.com/s/rqskio15lmxwo8c/dps%20psu%20box%20rev4.gcode?dl=0

(I used ideamaker 3 which is currently in RC stage - I'm not sure if 2.x is any different, but it looks like we're very close to the release of 3.x)

Related parts:
Generated by:
```
;Sliced by ideaMaker 3.0.5.1438, 2018-01-02 15:01:42 UTC+0100
```

Filament usage (at the end of the file, in mm):
```
;Material#1 Used: 13145.4
```
```
;Material#2 Used: 0.0
```

Layer height:
```
;LAYER:1
;HEIGHT:0.200
```

First layer height:
```
;LAYER:0
;HEIGHT:0.240
```

It seems to me that RRF gets first layer height from the actual code (correctly), and finding the layer height could be a bit difficult since it's multi-line. It can be worked around by adding this to the start g-code in ideamaker, emulating slic3r:
```
; layer_height = {layer_height}
```

Object height is also incorrect. It looks to me that RRF detects the last absolute Z movement, and sets that to object height. Unfortunately, ideamaker seem to raize Z by 5mm at the end of the print, still in absolute mode (`G0 F2400 Z70.040` in the attached code) - not sure how to handle this.

dc42, 2018-01-02:
You can get around the incorrect layer height determination if you can add a comment starting with ";E" or "; E" at the end of that G0 line.

Of the three methods that RRF uses to estimate print time, basing the estimate on number of layers completed is the least accurate. It's possibly not bad for a cube, but terrible on most real prints. The estimate based on filament consumption is usually the most accurate one.

andrasbiro, 2018-01-02:
Unfortunately it's some kind of auto-generated end code. If I could edit it, I would change it to relative.

But you're right, I originally started this ticket for the filament usage, and that would be still good enough.

Removing the layer height workaround would remove the layer time estimation as well, so it won't give false results on the web (which seem to calculate some kind of average)

dc42, 2018-01-10:
I'll add support for the filament usage comments in the next relea, assuming it doesn't turn out to be difficult. The layer height comment is trickier because RRF expects the layer height info to be on a single line; so I suggest you use the workaround. You might also like to suggest to the Ideamaker developers that they add a layer height comment at the start.

dc42, 2018-01-13:
Please make  GCode file available to me for testing with.

dc42, 2018-02-13:
Filament usage comments are recognised in the forthcoming 1.21RC2 release.

closed 2018-02-13 by dc42