Duet3D/RepRapFirmware #153: Feature Request: WPA2-Enterprise/PEAP support

pilotgary, 2018-01-03:
Our work WiFi has recently been reconfigured to use WPA(2)-Enterprise/PEAP instead of WPA2-PSK.  Could this be supported on the DuetWiFi?  I suspect it would involve too much work for the possible audience.

dc42, 2018-01-03:
I've done some research and it appears that the version 2.1.0 ESP8266 SDK that we use in DuetWiFiServer 1.20 does support that. But it would be necessary to extend the M587 command to configure at least the identification type and outer identity in addition to the existing parameters.

pilotgary, 2018-01-04:
Thanks for taking a look.  I'll leave it up to you, whether to implement it or not,  I'll be happy to test it, if you do.

Now that the LAN version of the Duet has silent stepping, I might go that route for a new CoreXY printer I'm building to use at work and take my delta home.

ccavanaugh, 2018-01-07:
This would be helpful as I am trying it integrate a Duet Wifi into a corporate network.  Flexibility of the network is limited so we are forced to use cellular hot spots or sd cards.

bluesign2k, 2018-01-07:
This would be helpful to me too. At the moment, we've been buying the Ethernet versions of the Duet purely because the WiFi version doesn't support this at the moment. However, I'm now running out of Ethernet sockets!

SijmenSchoon, 2018-01-07:
For my job, I've tried implementing support for eduroam on the ESP8266 (which is used for the DuetWifi's WiFi connection). Sadly however, the anonymous identity is hardcoded (in the closed-source part of their framework) to `anonymous@espressif.com`. Therefore does not work with WPA2-Enterprise that use this anonymous identity, which to my knowledge are most of them.

This means it won't be possible to *properly* support WPA2-Enterprise until either Espressif decides to implement setting the anonymous identity (which I don't deem as likely, knowing Espressif so far) or someone reverse engineers the WPA2 part of the framework (unlikely to be done in the foreseeable future too).

TL;DR: not possible until someone fixes Espressif's framework

dc42, 2018-01-07:
I think the Arduino ESP8266 thread I read said that problem had been fixed in the later SDK versions, by the addition of a new API call to set the outer ID. Or did I misinterpret it?

SijmenSchoon, 2018-01-07:
Oh, then they must have changed that. I'll check for you.

SijmenSchoon, 2018-01-07:
Anyway, I have a boring lecture tomorrow where I'll have access to eduroam. I could give implementing this a shot if you'd like - I have a few ESP8266's lying around anyway (and my DuetWifi is currently on its way)

mlbuxbaum, 2018-04-03:
DC42  is ther anyway to elaborate on the extended M587 Commands? I'm trying to get connected to our intranet wifi system at work and so far its a no go. Trying to exchange my wifi board for the ethernet version ......

T3P3, 2020-10-20:
this can be setup using a Duet connected to an SBC.

closed 2020-10-20 by T3P3