Duet3D/RepRapFirmware #154: Make it clear how to find the firmware!

dintid, 2018-01-04:
When going to https://github.com/dc42/RepRapFirmware there is no way to navigate to the newest firmware. Only option is to open EdgeRelease (and some other folders).
The only way to find the link to newest firmware is to google and then open https://duet3d.com/wiki/Where_to_get_Duet_firmware_and_tools and click the link to ->
https://github.com/dc42/RepRapFirmware/releases

Why isn't the folder "releases" available directly in the repository?

dc42, 2018-01-04:
It is. When you are at the root of the repository, you will see a line of links with these captions:

 976 commits    6 branches    6 releases    14 contributors    GPL-3.0

Click on 'releases'.

closed 2018-01-04 by dc42

dintid, 2018-01-04:
[image]
oh, I would never had figured that one. Maybe make a link to it as well.