Duet3D/RepRapFirmware #157: Feature Request: Turn off bed heaters for z-probing (BL-Touch specifically)

ka15574, 2018-01-31:
Been having some problems with the repeatability of my BL-Touch and stumbled across [this post on the Marlin repository](https://github.com/MarlinFirmware/Marlin/issues/6394). In short, people are having issues where the current through the heated bed is causing EMI through the probe tip that is interfering with the hall sensor and causing the measurements to vary. While the repeatability doesn't change drastically (from ~4 microns w/ bed off to ~10 microns w/bed on), for someone using bang-bang heating (like me) the effect of one probe position being measured with the bed on and the next with it off is sure to have an effect. A solution seems to be to turn off the bed when probing and then turn it on when moving to the next position, reducing the effect of the induced EMF but allowing the bed to be probed while hot. This seems easy to implement in principle but having limited coding skills I'm not sure where to start. Is this achievable?

lesaux, 2018-01-31:
you probably could add some commands to your retractprobe.g and deployprobe.g files to achieve this.

ka15574, 2018-01-31:
Makes sense, I'll take a stab at it when I get back to my house. I am concerned with maintaining the bed temp throughout the mesh bed leveling. I have the probing speed set to very low to get better results, but now this means that the off time for the bed is much longer than the on time per probing cycle. I'm predicting a significant net drop in temp by the time I get to the last point, which will definitely affect results more than the induced EMF. Preferably, the bed would only turn off _right_ before the probe hits, and then turn back on as soon as the distance is measured. I have no idea how I would implement that though.

eboston, 2018-01-31:
How about doing a set bed temp and wait, turn off bed, move to next point, repeat?

eboston, 2018-01-31:
Add a probe after turning off bed.

ka15574, 2018-01-31:
Just got back to my house, and the problem that I'm seeing is that the
deployprobe.g and retractprobe.g happen back to back right _after_ the
probing has been done, before the head moves to another spot. This means
that the time for which the bed would be off using those files would be
minuscule - a fraction of a second, leaving no space for user code to work.
I'm wondering if there's a way to incorporate it into the actual G29 code
in the firmware to turn off the bed heater right before probing and then
turn it on again before moving to the next position.

dc42, 2018-02-15:
Implemented in firmware 1.21RC2. Use parameter B1 in the M558 command to turn the heaters off during probing.

closed 2018-02-15 by dc42