Duet3D/RepRapFirmware #159: Changing firmware retract parameters while printing may result in print failure

FHeilmann, 2018-02-10:
The following happened to me yesterday:

I was printing a plate full of parts and eventually one of them came loose. To avoid having the nozzle get stuck on the slightly lifted part I wanted to change the firmware retraction z-hop from 0.2mm to 1mm.

Unfortunately, when I pressed enter in the gcode console and thus committed the new z lift, the printer was in the retracted state (z 0.2mm above the current layer height).

The firmware then undid the zlift using the new parameters and the nozzle ended up 0.8mm below the current layer height which destroyed the rest of the print and almost took my printhead to the grave in the process.

Now I know this is due to my own stupidity/bad luck, but It seems to me it would be easy (from a firmware perspective) to have the firmware always undo the amount it previously lifted instead of the amount commanded by the current state of the firmware retract parameters, and make it resilient to this type of user error in the process.

dc42, 2018-02-10:
The current code already does as you suggest. Which firmware version were you using?

FHeilmann, 2018-02-10:
1.20 stable

FHeilmann, 2018-02-13:
@dc42 can you reproduce this?

dc42, 2018-02-13:
I have reproduced it in the latest dev build. Will be fixed in next 1.21RC.

dc42, 2018-02-13:
Fixed in the forthcoming 1.21RC2 release.

closed 2018-02-13 by dc42