Duet3D/RepRapFirmware #160: 1.21RC2 / M558: new probe type P9/BLTouch, deploy/retract problem with multi-touch

hurzhurz, 2018-02-16:
It looks like automatic probe deploy and retract between each probing using the new probe type P9 doesn't work correctly when using multi-touch Z probing (parameter A).

```
M558 P9 X0 Y0 Z1 H5 F200 T6000 A5 S0.015
M557 X40:260 Y40:260 S55
```

Before probing the first point, deployprobe.g is run and after it triggered, retractprobe.g is run.
But before it tries to probe this point again, deployprobe.g is not run so that the head crashes into the bed.

When using default A1 / no multi-touch, deployprobe.g is correctly run before probing the next point.

I think this behaviour matches the description in the release notes that says, deploy should happen before a move that however doesn't happen for multi-touch probing.

dc42, 2018-02-17:
Was this with G29 probing, or bed.g probing, or both?

hurzhurz, 2018-02-17:
With G29, I didn't try bed.g(=G32?).

T3P3, 2020-10-20:
This should be working on later versions. please try the latest stable release and discuss on the forum if its not working for you.

closed 2020-10-20 by T3P3