Duet3D/RepRapFirmware #163: 1.21RC3 Prints do not start. Duet Ethernet.

Alfaa123, 2018-02-28:
Updated our testing machine to 1.21RC3 this morning. Unfortunately we didn't get far as we encountered a major bug in this release.

Upon starting a print, every heater starts to warm up. Once they get to proper temperature, they just... turn off again. No error messages or anything. Print stays at 0% complete indefinitely. Printer is not locked up, you can still pause and cancel the prints, the just never actually start.

 Downgraded immediately to 1.21RC2 and the problem went away.

dc42, 2018-02-28:
Did you amend the homing files as described in the upgrade notes, and check that homing works?

Alfaa123, 2018-02-28:
Aha, I got a little too excited and must have skipped over that part. I believe this is exactly what the issue is. Thank you!

closed 2018-02-28 by Alfaa123

dc42, 2018-02-28:
I'm glad you solved it; however there should have been an error message in the GCode Console page of DWC when the homing sequence was aborted.

Alfaa123, 2018-02-28:
Looking at the console on the PanelDue, an error was generated there, but it didn't get passed to DWC.

dc42, 2018-03-01:
Thanks, I'll test this. If the homing command came from the SD card print file then a message should have been written to both PanelDue and DWC. Whereas if you sent a homing command manually from PanelDue or DWC, the message would be written only to the device you sent the command from.