Duet3D/RepRapFirmware #165: Blinking Auto-Bed-Compensation button

Christoph13524, 2018-03-18:
Hi,

my Auto-Bed-Compensation button is blinking every 3 seconds in my macOS Safari browser (version 11.1) in the way the screenshot shows. It is not blinking in Chrome browser!
[image]

JohnOCFII, 2018-03-18:
I've also seen different behavior in Safari compared to Chrome on my Mac.  While I do most of my web browsing in Safari, I've decided to just use Chrome for DWC.

Kriechi, 2018-03-18:
This is the repository for the main firmware.
The issue tracker for DWC is over at https://github.com/chrishamm/DuetWebControl

/cc @chrishamm

Christoph13524, 2018-03-18:
Ah ok sorry, didn't know that! Next time I'll do it right.

closed 2018-05-25 by dc42