Duet3D/RepRapFirmware #166: Thoughts on moving to a CMake driven CLI build?

lylecheatham, 2018-03-20:
Hey there,

I'm just poking around with the firmware right now, and I can't help but notice that development is locked to a single IDE.  While I have nothing against IDEs in general, I think it's certainly cleaner to have the build be free running, as that way developers can choose their own IDE, or choose not to use one.  Would a transition like this interest the community?

crpalmer, 2018-03-31:
I would love that.

dc42, 2018-03-31:
The C and C++ code is independent of the build system, but the build configurations and compiler settings need to be stored somewhere. I choose to store them in Eclipse. Eclipse is open, cross-platform, and far more productive than using just a text editor.

You can already choose not to use an IDE, and a few people have done so. The .cproject file is in XML format, so you could write a XSLT script to export the project configurations and compiler settings to whatever format you need for CMake. If you write such a script then I will consider adding to to the project.

likha3d, 2020-06-16:
@dc42 I have some work supporting a CMake build, as I am trying to get RepRapFirmware onto another platform. Would you be open to accepting a PR supporting this? (CMake support would also have to be submitted for each of the dependencies of RepRapFirmware on other repo).

dc42, 2020-06-16:
I will continue to use Eclipse. I am open to PRs for supporting other build systems provided that they are not intrusive, however I wouldn't maintain the CMake support.

mfs12, 2021-07-29:
I build a first draft to integrate full support with cmake. Also I am willing to maintain it.

T3P3, 2023-05-04:
closed until there is an updated pull request for a CMAKE build system and an ongoing maintainer.