Duet3D/RepRapFirmware #167: Can't simulate unless machine is homed

ccavanaugh, 2018-03-25:
Attempts to simulate a print fail if the machine is not homed.

This occurs on a Duet 0.85 using the released 1.21 on a delta machine.

wilriker, 2018-05-03:
This can be worked around (or maybe this is the intended way to do it?!) by sending
`M564 H0` before starting a simulation that disables the move-without-homed-prevention.

You probably want to send `M564 H1` afterwards to reenable homing-requirement again.

See also #168

dc42, 2018-05-03:
This has been fixed in 2.0beta for the Duet 2. the fix will be included in the 1.21.1 release for the older Duets.

ccavanaugh, 2018-05-12:
I've test the 1.21.1RC with about a 1 hour print including the simulation and all appears well.

Thanks!!!!!

closed 2018-05-12 by ccavanaugh