Duet3D/RepRapFirmware #168: REQUEST: Option to disable the "must home axes" requirments

dintid, 2018-03-27:
The new **requirment to home axes** before being able to move anything might be a fine safety feature for some people, but it is unwanted in many situations, so please just add an option to disable this requirment.

mloidl, 2018-03-27:
This option is already present.
use M564 H0. see https://duet3d.dozuki.com/Wiki/Gcode#Section_M564_Limit_axes

dintid, 2018-03-27:
Thanks @mloidl Guess I can finally print something again now! Would be nice if changes this fundamental was better documented for people to be able to make the necessary adjustments.

dintid, 2018-03-27:
@mloidl Do you have any idea why it decides to home to U max (my virtual z-max) between each G30 probe as defined in bed.g?
I'm using a 3-point home sequence for true autolevel as I have 3 independent motors. 
The U axis is a way to make it home to Z-max, but I have no idea why it started going to U-max between each probe?!

mloidl, 2018-03-27:
Unfortunately not. I've just managed to move all axes and home x/y on my HEVO.
I think for these kind of questions it would be better to ask in the forum: https://www.duet3d.com/forum/

dintid, 2018-03-27:
Sure, I just thought you might know it :)