Duet3D/RepRapFirmware #169: Can't print without start.g in place - readme is wrong.

dintid, 2018-03-27:
In the readme it states we must remove start.g file if we have it in place, but the printer tries to load it up before each print and fails if no file is in place.
Creating an empty file fixes this, but either the notes are wrong, or the checking on the machine is.

dc42, 2018-03-27:
Please explain what you mean by "fails". 2 of my 3 printers have no start.g file and I do prints on them.

dintid, 2018-03-27:
Sure. When I start a print without any start.g in place it fails with this errorcode:
**Error: Can't open 0:/sys/start.g to read, error code 4**
When I create an empty start.g it works perfectly.

dc42, 2018-03-27:
Looks like you are running with debug enabled. If you disable debugging then there will be no message.

dintid, 2018-03-28:
I run debug enabled yes, so I get a message telling me why my printer stops working whenever I update firmware... 
The problem is not the errormessage itself, but the issue that the printer don't start printing unless I create this file, which you explicitly write we should delete if in place.

dc42, 2018-03-28:
1. You do not need to create this file for the printer to print.
2. You should never normally enable debug during printing. Enabling debugging will slow down the printer, especially if you don't have a PC connected via USB because then there is nothing to receive the debug data.

closed 2018-03-28 by dc42

apphead, 2018-09-28:
> 1. You do not need to create this file for the printer to print.
> 2. You should never normally enable debug during printing. Enabling debugging will slow down the printer, especially if you don't have a PC connected via USB because then there is nothing to receive the debug data.

Both 1. and 2. are true. But why does the firmware call start.g in the first place if it is not required?
In my understanding, the debug flag should be used for troubleshooting, and if it reports a problem such as the missing start.g file, then it is an error that should be fixed. Or is it a backward compatibility issue?

dintid, 2018-09-28:
> 1. You do not need to create this file for the printer to print.
> 2. You should never normally enable debug during printing. Enabling debugging will slow down the printer, especially if you don't have a PC connected via USB because then there is nothing to receive the debug data.

Actually we do need to create this file to print, if debug is enabled, as debug (I guess) stops the printer from starting if it's not in place.
This is not a normal behaviour of a debugging which should flag and log the error issue, not in any way take action based on the results.
If the printer acts differently when debugging is enabled versus when it's no enabled, aside from possible speed issues (which I've never encountered), then the debug itself is really not very usefull.
Observing (debugging) someting should not affect the results and certainly not cause the printer to behave differently.

It's a problem how every issue that breaks the printer after an update is after rationalized instead of taken seriously.
It would be very simple to just clearly list these 1-2 printer breaking changes each firmware update, instead of hiding them in the huge list, which is irrelevant for most people.

Most people update to get better stability and/or performance. Every now and then users gets new functions, which is great, but making changes that stops existing setups from functioning and not making this absolutely clear (no, hiding them in the long changelog is not making it clear), is really your number 1 problem as a company seen from the user perspective.

Would be like changing a router from going IPv4 and IPv6 to only IPv6 because of performance issues, but not making a big fat warning at the top of the list and when downloading, that precation 1, 2 and 3 should be taken, as this might make the router defunct depending on the ISP in question.

dc42, 2018-09-29:
If you enable any sort of debug, **you must have a device connected to the USB port to receive the debug information**. This is not specific to having a start.g file or not, it applies to **any** debug message.

You still haven't said what goes wrong when debugging is enabled and no start.g file is present.  The only thing I can see that is wrong is that the message should not be prefixed with "Error" because it is quite a normal occurrence for an optional macro file to be missing. I'll change that.