Duet3D/RepRapFirmware #170: 1.21 Print always fails before starting if axes aren't pre-homed.

dintid, 2018-03-27:
After update, using cura as slizer.
Print always fails to start unless all axes are pre-homed.
I have G28 XY in startup, then G28 U (which corresponds to Z-max) then alter G32.. but it fails before it even gets to use G28 with below error:
```
07.27.13Heater 1 switched off
Heater 3 switched off
Cancelled printing file C_bottom_shell_ZeroPro.gcode, print time was 0h 1m
Error: G0/G1: insufficient axes homed

07.26.35Heater 1 switched on
```

dc42, 2018-03-27:
Read the version 1.21 upgrade notes and fix your homing files.

dintid, 2018-03-28:
I have read all notes, but nothing said anything about special requirments in homing files?!
What if you read what I write, or just explains what goes on, instead of tossing the ball like that?

My homing files works fine.. I said it worked if I PRE-HOMED. Meaning, I homed axes before starting print... so I can home, but not when the same files are called in cura startup gcodes...

I HAVE read it.... I really hate the way you just say "read the notes or wiki" instead of expecting that we did - I don't expect that's the case, but you sound extremely arrogant and not helpfull at all, when you always do this... 
...the (smaller) issue here is not getting what you are writing, not that people don't read it. Most people read these things before asking for help.. you should know I do... but you STILL tell people to read them again... 

**It would be nice if special focus were made to explain these machine breaking changes.. and make them optional, instead of forcing it down on us, and make it possible to disable it.**

**If unclear, I don't know what needs to be fixed in homing files** as I don't know how I can home my axes if no moves can be made before I've homed my axes..

dc42, 2018-03-28:
1. The problem you are having is because you are missing the S2 parameter in some of the movement commands in your homing files. This is explained in the upgrade notes.

2. I **have** made it possible to disable this change, that's what M564 H0 is for. So I'm sorry, I really don't see what you are complaining about. The default behaviour will always be the safe one. RRF is increasingly used in laser cutters and CNC machines where the consequences of movement or other actions before homing can be severe. Also, several users have complained to me that they have damaged something because movement was allowed before homing.

3. Explaining breaking changes is what the upgrade notes are primarily for, and that is why they are always at the start of the section on a new release in the whatsnew file.

4. Sadly neither me nor the earlier developers of RRF always had the foresight to get all aspects of machine behaviour exactly right, hence the need to make breaking changes to correct errors that were made previously. As the rate of adoption of RRF is increasing rapidly, IMO it is worth causing existing users the inconvenience of making changes to config and homing files in order to spare future users from the effects of bad design decisions made in the past.

CraigMason, 2018-04-12:
Hi @dc42 

Just adding my comments as I upgraded to 1.21 last night and experienced a somewhat related issue (which this thread helped me resolved)

I don't have a level sensor as I have a very flat bed and the manual probing with a 0.04mm feeler guage works very well. However, it was not possible to use manual bed probing after the update as the G0/G1 commands issues from the pop-up aren't usable without already being homed.

I agree that not allowing movement for un-homed axis makes a lot of sense. Perhaps the addition of a sentence in the 'upgrade notes' regarding manual bed probing and the need for `M564 H0` would be of benefit.

Thank you!

dc42, 2018-04-12:
Thanks Craig, we'll fix that in a future release.

dc42, 2018-04-28:
I have put a fix for this in the forthcoming 2.0beta3 release of RRF.

ThorsenRune, 2018-09-24:
Hi, sorry to disturb.
I upgraded my ormerod 1 RepRapPro with dual extrusion extension to RepRapFirmware for Duet FIRMWARE_VERSION: 1.21RC3 and face the homing issue.
My z-sensor is not doing a good job so I prefer manual homing. 
However the M564 H0 mentioned above does not seem to remove the "G0/G1: insufficient axes homed" message for me.
Anyway to simply disable this?
Should I just send a G92  Z0 command to simulate that it has been homed?
Thanks for your help
Rune
P.S. Would it be worth upgrading with such an 'old' printer?

dc42, 2018-09-24:
You are using an obsolete release candidate. Please upgrade to a later firmware version, see https://github.com/dc42/RepRapFirmware/releases/tag/2.01. Use the forum at duet3d.com for support.

closed 2018-09-24 by dc42