Duet3D/RepRapFirmware #183: CoreXY issue when using G1 S2 X10 due to Not being able to move without homing

Papsnoek, 2018-06-24:
Hi Guys,

I see two other issues around the Moving while not Homed that was introduced for CoreXY. While I understand this is a good thing to implement, it seems when using the G1 With the S2 command to move while not homed causes the kinematics to move like a Cartesian Printer and no longer like a CoreXY Printer.

This is on the latest release 2.01Beta1. I first picked this up when trying to move my axis with the Web Control. I then used the GCode console to do a Move G1 X10 without the S2 and everything works perfect (After adding the M564 H0 command to the config.g file). 

Thanks for an awesome probuct by the way :)

T3P3, 2020-10-20:
If this is still an issue please discuss the requirement on the forum.

closed 2020-10-20 by T3P3

dc42, 2020-10-20:
Note: G1 H2 (previously G1 S2) moves are **defined** to be individual motor moves.