Duet3D/RepRapFirmware #185: WS2811/WS2812 LED strip control

Armadillo1, 2018-06-25:
Would be a useful feature to be able to tell from a distance whether the hotend is cold (e.g. green), heating (yellow) or hot (red). 
An option for ambient (low light), working lights when repairing the printer (=bright), etc would be nice and helpful.

Something to consider.
Thanks

lordgort, 2018-07-20:
yes yes yes XD this would be awesome

lordgort, 2018-07-20:
but i think this is more a request, not an issue

dc42, 2018-07-20:
We will definitely not be supporting the WS2811/WS2812. The reason is that they use a nonstandard protocol that means the processor cannot do anything else (even process interrupts) while writing to these LEDs. That is unacceptable; it would lead to lost steps, lost serial input, possibly also network disconnections.

However, we have implemented a driver for DotStar LED strips in the latest RRF source code. These LED strips use standard SPI MOSI ands SCLK signals, so not only does the processor not need to give them its undivided attention, it can use DMA to send the data. To use them on the Duet WiFi or Duet Ethernet, you would need to make a hardware adapter to level shift the MOSI and SCLK signals to 5V, also to gate SCLK with your chosen /CS pin. You could do all of this using a single 74HCT02 chip. You would also need to enable the DotStar support for the Duet 2 build of RRF.

closed 2019-01-22 by dc42

svs072477, 2019-04-10:
Studying the topic with WS2812 came across two projects using DMA, perhaps it will be useful.
https://www.pjrc.com/non-blocking-ws2812-led-library/
https://drive.google.com/file/d/0B5dbvc_yPqJHQ2FEUXpkR3NocnM/view?usp=sharing

dc42, 2019-04-10:
It may well be possible, but it is inefficient, and the timing accuracy would depend on what other DMA is taking place at the same time. Both the network interface SPI and the SD card use a lot of bandwidth, e.g. 20MBytes/sec for the SD card. DotStar is a better solution.

svs072477, 2019-04-10:
DotStar of course better. And have you had experience with other SPI LED strips? For example APS 102 https://cdn-shop.adafruit.com/datasheets/APA102.pdf In Russia, they are easier to buy than Dotstar

svs072477, 2019-04-10:
I realized that asked a stupid :)

dc42, 2019-04-10:
You mean because the DotStar originally used the APA102? That's my reading of the tech details section of https://www.adafruit.com/product/2238?length=1.

svs072477, 2019-04-10:
Yes, Yes, I already wrote above that asked a stupid question :)

stavros-k, 2019-04-11:
If you have a lot of ws2812 hanging around and you don't want get apa102 or other supported led strips. You can get an arduino (as a cheap as 2$) and connect it with duet wifi to get the status and control the leds.
Duet <-> arduino - > leds