Duet3D/RepRapFirmware #187: Can not find FreeRTOSConfig.h

tobbelobb, 2018-06-29:
Hi dc42!

I've started adding Hangprinter to RRF, and am about to compile my first binary with Eclipse.

Since I know RRF will soon use FreeRTOS (announcement https://reprap.org/forum/read.php?146,818099), I want to compile with Configuration SAM4E8E_RTOS in my C/C++ Build Settings. However, I can't find RRF's `FreeRTOSConfig.h` to complete the build. It it published?

tobbelobb, 2018-06-29:
I was able to find a `FreeRTOSConfig.h` here:
https://github.com/dc42/FreeRTOS

If this indeed corresponds to the RRF code then it should be part of the RRF source code, not the FreeRTOS source (see page 11 in FreeRTOS reference manual).

I've guessed that it's supposed to be copied to the eclipse workspace, judging from what I found here:
[image]

I've also imported the (master branch of dc42's repo of) FreeRTOS source into Eclipse. That takes me to a very similar error message as in #188. I'm probably missing something in build environment, or are linking something in the wrong order...

I'm using Ubuntu18.04. I've tried installing everything from the apt repos named newlib- something arm- something:
```
sudo apt install libnewlib-dev libstdc++-arm-none-eabi-newlib newlib-source libnewlib-arm-none-eabi
```
... but it doesn't seem to make any difference.

I also get a lot of conflicting CPU architectures errors like this:
```
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/crt0.o: Conflicting CPU architectures 13/1
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/crt0.o
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: Duet2CombinedFirmware.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libm.a(lib_a-s_llrint.o) does not
```
Full console output here (errors start on line 105):
https://pastebin.com/sT1YZVVE

tobbelobb, 2018-06-29:
I was able to get around the issue by installing the toolchain in another way. I followed the instructions for manual GNU/Linux installation, found here:
https://gnu-mcu-eclipse.github.io/toolchain/arm/install/

I then set the cross settings like so:
[image]

andreq, 2018-07-07:
I just got latest v2 dev branch and had similar issue with those libs.

To compile RRF you need a really specific folder structure 

Root > 
      "FreeRTOS"  > drop FreeRTOS repo in here, all the files
      "CoreNG"  >  drop v2 branch of CoreNG repo in there, all the files
      RRF > I  got my git clone of RRF repo in there, forlder name isn't trivial for this one.

Name in " " have to be exactly named like this.

In eclipse, you need to import each Project folder into your workspace. The base RRF project config should then link all that stuff properly. 

I had to change my GCC path in the C/C++ build variable, but that's it.

To compile you first Build CoreNG (SAM4E8E_RTOS for duet 2 rtos)
Clean RRF project
Build RRF project (Duet2_RTOS)

dc42, 2018-07-07:
Most of that was already covered in the  BuildInstructions.md file. I have added step 5b to cover adding the FreeRTOS project when using the v2-dev branch.

tobbelobb, 2018-07-07:
There are no instructions for Ubuntu 18.04 GNU/Linux in BuildInstructions.md.

Gist of this thread:
 - Ubuntu 18.04's repos does not contain sufficient versions of eclipse, eclipse-cdk, and the ARM tool chain. They must all be installed manually (see above).
 - RRF's `FreeRTOSConfig.h` is placed in a fork of the FreeRTOS source. Best practice is to have `FreeRTOSConfig.h` be part of the application source.

Thanks for extending BuildInstructions.md. I will help those who come after me (future RRF newcomers) in another way.

#### Hangprinter Assembly Workshop Live OS (HAWs)
Each time I give a Hangprinter assembly workshop, I include a USB stick with a live OS on it. The OS contains a fully configured development environment for all code/design files relevant to Hangprinter. I will include eclipse, RRF, and all dependencies in the OS as soon as the next assembly workshop get booked. 

This saves us from having to learn and document quirks of other OSes. Build/installation instructions get reduced to: Download HAWs, boot HAWs, start Eclipse, press "build". Bugs and error messages get reproducible, and only one version of any library need to be supported.

When new developers want to configure the tool chain on their non-live OS, they know what the GUIs look like and how they should behave. Configuration files can be copied directly from the USB into their own system.

More on HAWs in [this blog post](http://vitana.se/opr3d/tbear/2018#hangprinter_project_57).

BikingViking77, 2018-07-07:
Just in the process of setting up an Ubuntu dev machine as it happens, appreciate all the hints!

dc42, 2018-07-12:
I welcome your initiative to provide an environment to make it easy to build RRF under Ubuntu. However, we cannot be constrained by the versions of Eclipse, gcc etc. that are available in a particular Ubuntu repository where they are not the latest release versions.

The FreeRTOSConfig.h file needs to be part of the FreeRTOS project because FreeRTOS is built as a separate library that is used by RepRapFirmware and other projects. This would not be manageable if the FreeRTOSConfig.h file were within the application source tree. Different FreeRTOS builds are handled by the project settings.

closed 2019-01-22 by dc42