Duet3D/RepRapFirmware #192: 1.21 M564 H0 does not work with radds

clearbucketLabs, 2018-07-11:
I have a working config from 1.19, updated to 1.21 and added M564 H0 to the config.  even if send it manually i still get "insufficient axes homed".

actually i can't move anything at all... even with G0 Zxx S2.... no error but Zero movement.

dc42, 2018-07-12:
I can see no reason for this behaviour, assuming you really are running 1.21 and not an earlier beta of version 1.21. Please re-test it. Also, try sending M564 without parameters to check that your M564 command in config.g was processed properly.

clearbucketLabs, 2018-07-16:
Hey!  your right, looks like it was an RC3 or something.  after painfully re-flashing (hitting the erase) it seems good.

closed 2018-07-16 by clearbucketLabs