Duet3D/RepRapFirmware #195: Mixed up heater mapping with IDEX and 4 extruders,hotends

thehans, 2018-08-01:
I'm attempting to configure Duet 2 Wifi , with DueX5 in preparation for a large printer build which will be a cartesian configuration with dual X carriages, and 2 extruders on each carriage for a total of 4.

I have updated to the latest firmware 2.01(RTOS), and am using stacked dual PT100 interface boards for sensors. The electronics are able to read the temperature for all 4 sensors correctly.  However I'm having issue with defining the tool heads and getting the correct heater to correspond with the tool.

With the configuration below, the LCD is showing only 2 of the tools.  And If I try to turn on the first nozzle, just to a low 40C temp for testing, it starts to heat the correct hotend, but its acting like it thinks it is heating the 3rd one(temp highlighted red on LCD) and will thermally run away far beyond the set point on the first hotend.

[image]

The setup is currently just being tested on the bench, printer mechanicals have not been completed yet for full integration testing.  I have stepper motors connected to drivers for X/Y/Z/U , but none yet connected for extruder motors.  (I don't think this should affect the toolhead definitions though?)

My plan was to run all 4 heaters and extruder steppers off of the DueX5 board just for wiring neatness and consistency, and ignoring the 2 heaters on the main board.  Any help is appreciated.

config.g shown below:
```
; Configuration file for Duet WiFi (firmware version 1.21)
; executed by the firmware on start-up
;
; generated by RepRapFirmware Configuration Tool on Tue Jul 31 2018 14:57:20 GMT-0500 (Central Daylight Time)

; General preferences
G90                                        ; Send absolute coordinates...
M83                                        ; ...but relative extruder moves

; Network
M550 PFourHead                             ; Set machine name
M552 S1                                    ; Enable network
M586 P0 S1                                 ; Enable HTTP
M586 P1 S0                                 ; Disable FTP
M586 P2 S0                                 ; Disable Telnet

; Drives
M569 P0 S1                                 ; Drive 0 goes forwards
M569 P1 S1                                 ; Drive 1 goes forwards
M569 P2 S1                                 ; Drive 2 goes forwards
M569 P3 S1                                 ; Drive 3 goes forwards
M569 P4 S1                                 ; Drive 4 goes forwards
M569 P5 S1                                 ; Drive 5 goes forwards
M569 P6 S1                                 ; Drive 6 goes forwards
; Secondary X axis is labeled "U"
M584 X0 Y1 Z2 U3 E5:6:7:8                     ; Apply custom drive mapping
M350 X16 Y16 Z16 U16 E16:16:16:16 I1           ; Configure microstepping with interpolation
M92 X80 Y80 Z4000 U80 E420:420:420:420         ; Set steps per mm
M566 X900 Y900 Z12 U900 E120:120:120:120        ; Set maximum instantaneous speed changes (mm/min)
M203 X6000 Y6000 Z180 U6000 E1200:1200:1200:1200 ; Set maximum speeds (mm/min)
M201 X500 Y20 Z250 U500 E250:250:250:250        ; Set accelerations (mm/s^2)
M906 X800 Y800 Z800 U800 E800:800:800:800 I30   ; Set motor currents (mA) and motor idle factor in per cent
M84 S30                                    ; Set idle timeout

; Axis Limits
M208 X0 Y0 Z0 U0 S1                           ; Set axis minima
M208 X500 Y700 Z1000 U500 S0                    ; Set axis maxima

; Endstops
M574 X1 Y1 Z1 U2 S1                           ; Set active high endstops

; Z-Probe
M558 P1 H5 F120 T6000                      ; Set Z probe type to unmodulated and the dive height + speeds
G31 P500 X0 Y0 Z2.5                        ; Set Z probe trigger value, offset and trigger height
M557 X15:485 Y15:685 S20                   ; Define mesh grid

; Heaters
M305 P0 T100000 B4138 C0 R4700             ; Set thermistor + ADC parameters for heater 0
M143 H0 S120                               ; Set temperature limit for heater 0 to 120C

M305 P1 X200
M143 H3 S280
M305 P2 X201
M143 H4 S280
M305 P3 X202
M143 H5 S280
M305 P4 X203
M143 H6 S280

; Fans
M106 P0 S0.3 I0 F500 H-1                   ; Set fan 0 value, PWM signal inversion and frequency. Thermostatic control is turned off
M106 P1 S1 I0 F500 H1:2:3:4 T45            ; Set fan 1 value, PWM signal inversion and frequency. Thermostatic control is turned on
M106 P2 S1 I0 F500 H1:2:3:4 T45            ; Set fan 2 value, PWM signal inversion and frequency. Thermostatic control is turned on

; Tools
; Pn is tool number, not heater
; H0 is for bed, skip H1 and H2 on main board, H3 is first heater on DueX5?
; D0 is first extruder driver, skip 2 on main board, so D2 is first extruder driver on DueX5?
M563 P0 D2 H3                             ; Define tool 0
G10 P0 X0 Y40 Z0                           ; Set tool 0 axis offsets
G10 P0 R0 S0                               ; Set initial tool 0 active and standby temperatures to 0C
M563 P1 D3 H4                              ; Define tool 1
G10 P1 X0 Y40 Z0                           ; Set tool 1 axis offsets
G10 P1 R0 S0                               ; Set initial tool 1 active and standby temperatures to 0C
M563 P2 D4 H5                              ; Define tool 2
G10 P2 X0 Y-40 Z0                          ; Set tool 2 axis offsets
G10 P2 R0 S0                               ; Set initial tool 2 active and standby temperatures to 0C
M563 P3 D5 H6                              ; Define tool 3
G10 P3 X0 Y-40 Z0                          ; Set tool 3 axis offsets
G10 P3 R0 S0                               ; Set initial tool 3 active and standby temperatures to 0C

; Automatic saving after power loss is not enabled

; Custom settings are not configured
```

dc42, 2018-08-01:
Please ask support questions on the forum at https://forum.duet3d.com/, ont here.

thehans, 2018-08-01:
OK, my apologies, it seemed like there was a bug involved between which heater it turns on vs which one it monitors, but its entirely possible that I just don't understand the numbering or something.  I'll post this to the forums.

closed 2019-01-13 by thehans