Duet3D/RepRapFirmware #196: RADDS max end stops

CthulhuLabs, 2018-08-05:
David,

On the RADDS board only the min stops are defiined. This:

http://doku.radds.org/wp-content/uploads/2018/02/RADDS_DUE_PINS.png

shows the endstop pins to be:

X_min 28
Y_min 30
Z_min 32
X_max 34
Y_max 36
Z_max 38

Can you please add this in your next point release? I have a z max endstop and an trying to probe the bed with your z probe.

CthulhuLabs, 2018-08-05:
Also why is pwm3 listed as an endstop? Seems weird to me.

apphead, 2018-09-29:
Is RADDS supported by dc42 fork?

dc42, 2018-09-30:
See section "Supported hardware" at https://github.com/dc42/RepRapFirmware/blob/dev/README.md.

dc42, 2018-09-30:
> Also why is pwm3 listed as an endstop? Seems weird to me.

I don't know. I didn't do the RADDS port.

closed 2020-10-20 by T3P3