Duet3D/RepRapFirmware #199: Getting recurring error on latest firmware - Error: M32: Cannot set file to print, because a file is already being printed

jgrouse, 2018-08-14:
After updating the firmware, if I inadvertently cause an error such as starting a job after one is already started, cancel the first job and then start a new one, it appears to not clear the first error and keeps logging the following message every 15 sec while the new job is running.  
 
File xxxxx.gcode selected for printing
Error: M32: Cannot set file to print, because a file is already being printed
Resume-after-power-fail state saved
Error: Attempting to extrude with no tool selected.
Printing paused
Cancelled printing file xxxxxx.gcode, print time was 0h 1m

As soon as the job finishes it stops logging these however they come back when I start the next job.  It's like it is not clearing those errors after showing them for the first time.

T3P3, 2018-08-14:
please can you use the duet 3d forum for support:
https://forum.duet3d.com/category/6/firmware-installation

dc42, 2018-08-14:
And please state which firmware version you are using in your forum post.

closed 2018-08-27 by dc42